TheThingsNetwork/lorawan-stack #1514: Getting error when executing instructions to run things stack on ubuntu

emilppeter, 2019-10-25:
Hi, 
I was actually following the getting started instructions to run things stack on ubuntu 18.04 . When I tried executing the following instruction: 
docker-compose run --rm stack is-db create-admin-user \
  --id admin \
  --email your@email.com 
I got the following error:
error:cmd/ttn-lw-stack/commands:missing_flag (missing CLI flag `email`)
    flag=email
Any help would be appreciated.
Thanks in advance.

rvolosatovs, 2019-10-25:
It sounds like you forgot the `\` at the end of the line.
The command should look like this:
```bash
docker-compose run --rm stack is-db create-admin-user \
  --id admin \
  --email your@email.com
```

Alternatively you can just put everything on one line and that should work fine:
```bash
docker-compose run --rm stack is-db create-admin-user --id admin --email your@email.com
```

Please let me know if this helps!

emilppeter, 2019-10-25:
Thanks @rvolosatovs ! It worked for me.

closed 2019-10-25 by emilppeter