TheThingsNetwork/lorawan-stack #1521: Editing unexposed root keys

johanstokking, 2019-10-30:
#### Summary

Support editing the device on a Join Server while not exposing root keys.

#### Why do we need this?

To avoid having root keys generated while the root keys are actually not exposed by the Join Server.

#### What is already there? What do you see now?

When editing a device with its root keys not exposed (i.e. `root_keys != nil`, but `root_keys.nwk_key == nil` or `root_keys.app_key == nil`), the Console shows empty input boxes for the root keys. When saving the device, root keys get generated, essentially overwriting the unexposed root keys in JS.

Note that unexposed root keys are basically root keys available to JS, but not returned by the device registry. The difference here is `root_keys == nil` vs `root_keys != nil` with individual key fields `nil`.

#### What is missing? What do you want to see?

If JS doesn't expose a root key, do not require the user to enter a root key or automatically generate a root key if the box is kept empty. If the root keys are not exposed, the Console should by default disable the input box and not include the root key in the field mask.

#### How do you propose to implement this?

Probably easiest is to have checkboxes near the root key input boxes, to indicate whether you want to set or generate root keys;

- Checked: current behavior; show root key if present. On save, set root key from box or generate if box is empty
- Unchecked: do not touch the root key, i.e. do not pass `root_keys.{nwk|app}_key` in field mask

#### Can you do this yourself and submit a Pull Request?

Can review

johanstokking, 2019-10-30:
Refs #1473

johanstokking, 2019-10-30:
See also detecting whether keys are exposed or not in https://github.com/TheThingsNetwork/lorawan-stack/pull/1475

kschiffer, 2019-11-01:
Would the following work:

- We add a generator button to the root key inputs
- We make the root key inputs required (if not external join server)
- In the update form, when root keys are not exposed:
  - root key inputs will be disabled 
  - No root key will be generated automatically
  - No root key info will be sent to the registry
- Otherwise:
  - root keys can be changed (via the generator button or manual input)

johanstokking, 2019-11-01:
Sounds good to me!

Just out of curiosity; the create vs update behavior you describe is not related, right? The former improves the UX, the latter fixes this issue, right?

PR #1539 (2019-11-04): Fix unwanted overwrite of unexposed root keys in (device settings)
#### Summary
Closes #1521 

#### Changes
- Extend validation schema to not generate the random value when the root keys are unexposed
- Disable the input when the root keys are unexposed
- Add an input placeholder to inform about unexposed keys

#### Notes for Reviewers
@johanstokking I tried to keep the UX as it was (generating root keys when field is empty) for consistency. Only difference now is that the input will be disabled and no keys will be generated when root keys are unexposed.

#### Checklist
- [x] Scope: The refere [... cut]
closed 2019-11-04 by johanstokking