TheThingsNetwork/lorawan-stack #1538: Unable to login using ttn-lw-cli login command

emilppeter, 2019-11-01:
Hi ,
I am trying to start up things network stack using the github repository on my ubuntu environment following the instructions on this link https://github.com/TheThingsNetwork/lorawan-stack/blob/master/DEVELOPMENT.md. So after building ttn-lw-cli, when I try to do a “go run ./cmd/ttn-lw-cli login” I get the following error:
ERROR Please login with the login command
INFO Opening your browser on https://localhost:8885/oauth/authorize?client_id=cli&redirect_uri=local-callback&response_type=code
INFO After logging in and authorizing the CLI, we’ll get an access token for future commands.
INFO Waiting for your authorization…
The webpage pops up but it doesn’t load. Could you please help me out on it? Please let me know what I am doing wrong. Thanks in advance.
Emil Peter

rvolosatovs, 2019-11-04:
Have you successfully built the frontend before trying to login?
> mage js:build

emilppeter, 2019-11-05:
Thanks for replying. Yes I have successfully rebuilt the frontend.

johanstokking, 2019-11-05:
Does it work now?

The order is `mage js:build`, then `go run ./cmd/ttn-lw-stack start`, then `go run ./cmd/ttn-lw-cli login` in a different terminal.

emilppeter, 2019-11-06:
Thanks @johanstokking . It worked!!

emilppeter, 2019-11-06:
I am sorry. After logging in and proceeding further, when I try to execute the following command as described in getting started document: go run ./cmd/ttn-lw-cli gateways list-frequency-plans,I get the following error:
WARN grpc: addrConn.createTransport failed to connect to {localhost:8884 0  <nil>}. Err :connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority". Reconnecting...
Please help me out on this.Thanks in advance.

johanstokking, 2019-11-23:
@emilppeter this is a separate issue. Please open a new issue.

To fix this, please set `TTN_LW_CA=cert.pem` in env or pass `--ca cert.pem` (or at least what you have configured as self-signed cert)

closed 2019-11-23 by johanstokking