TheThingsNetwork/lorawan-stack #1552: Compute nsScheduleWindow at runtime

rvolosatovs, 2019-11-05:
#### Summary

`nsScheduleWindow` can be computed at runtime

#### Why do we need this?

To allow more time for AS to update the queue on uplinks, for example, and generally improve performance and throughput.

#### What is already there? What do you see now?

`nsScheduleWindow` hard-coded to a particular value(since #1551 it will be equal to 100ms)

#### What is missing? What do you want to see?

`nsScheduleWindow` computed at runtime.

#### How do you propose to implement this?

Each time processing a downlink task results in scheduling attempt on GS, NS can compute the difference between scheduled starting time of the task and the time response from GS is received. Use moving average(e.g. EMA) of the difference of these 2 timestamps as the `nsScheduleWindow` value.

#### Can you do this yourself and submit a Pull Request?

yes

rvolosatovs, 2020-02-11:
Using 99th percentile is definitely better approach here

johanstokking, 2020-05-08:
Is this still relevant since #2418 ?

rvolosatovs, 2020-05-08:
> Is this still relevant since #2418 ?

Yes

johanstokking, 2020-05-09:
> Each time processing a downlink task results in scheduling attempt on GS, NS can compute the difference between scheduled starting time of the task and the time response from GS is received.

Few things to consider here;
1. Not all downlink traffic goes to a cluster-local GS. We can only do this for downlink to a cluster-local GS target
2. The response time of GS doesn't say anything
3. The reported `delay` in the GS response is partly based on gateway round-trip times and partly on duty-cycle constraints in non-time-bound downlink (class C)
4. What you need here is the round-trip time from NS to gateways that are in the cluster. This needs reliable round-trip time observation by GS and the API needs to accommodate feeding this information to NS

rvolosatovs, 2020-05-11:
> starting time of the task and the time response from GS is received.

This is incorrect, `nsScheduleDelay` is difference between the time downlink task is scheduled for and the time downlink task was actually executed at.
If task is scheduled for time `T`, but it executes at `T+X`, `nsScheduleDelay == X`