TheThingsNetwork/lorawan-stack #1564: Combine multiple fetching processes during initial site setup into one

kschiffer, 2019-11-06:
#### Summary
Consecutive fetching processes in the console during initial load should be combined in a way that the user only sees one spinner for the entire loading sequence.

#### Why do we need this?
The experience on page load is currently quite disruptive. One can see multiple loading spinners appearing and disappearing until all page parts have been loaded. It would be better to display a single loading process. 

#### What is already there? What do you see now?
Spinners being displayed during different fetching processes.

#### What is missing? What do you want to see?
A way to show only one loading spinner until all necessary fetches to display the page have been made.

#### Environment
Browser.

#### How do you propose to implement this?
I think this could be quite challenging to pull off but would be interested in your opinion @bafonins.

#### Can you do this yourself and submit a Pull Request?
If we can find a way to implement this, then yes.

bafonins, 2019-11-07:
What pages in the console suffer from this?

kschiffer, 2019-11-07:
About any page, but most noticeable on device subpages, e.g. device overview.
On that page, you can see three separate loading spinners appearing consecutively.

kschiffer, 2021-08-02:
I don't deem this necessary anymore.

closed 2021-08-02 by kschiffer