TheThingsNetwork/lorawan-stack #1567: Add more fine-grained feature toggles e.g. for JS only deployments

kschiffer, 2019-11-07:
#### Summary
We need more fine-grained feature toggles for JS only deployments to ensure proper UX for the console on such deployments.

#### Why do we need this?
On JS only deployments, there will be a lot of functionalities not working in the console, so these should be properly hidden or disabled.

#### What is already there? What do you see now?
Feature toggles via #1543 

#### What is missing? What do you want to see?
More fine-grained toggles for JS only deployments (and maybe other special stack component set ups)

#### Environment
Browser.

#### How do you propose to implement this?
Once I know the exact requirements for individual features of the console, it is easy to add these to the feature checks. @htdvisser, please assist me with defining these requirements.

#### Can you do this yourself and submit a Pull Request?
Yes.

htdvisser, 2019-11-14:
With the latest additions in https://github.com/TheThingsNetwork/lorawan-stack/issues/579 I think this issue is now also covered.

rvolosatovs, 2020-03-12:
See https://github.com/TheThingsNetwork/lorawan-stack/issues/2119#issuecomment-597736420
> It does not make sense to allow creating ABP devices here, nor any NS/AS settings (MAC versions, payload formats, etc), nor choosing for "external JS" (because this JS is already "external" because there is no NS and AS enabled).

johanstokking, 2020-03-24:
@laurensslats can you describe the user flow here?

bafonins, 2020-06-23:
I checked various parts of the Console in IS+JS only setup. The only problem I was able to find (having https://github.com/TheThingsNetwork/lorawan-stack/pull/2644 in place) is that the users are not able to register new gateways. https://github.com/TheThingsNetwork/lorawan-stack/pull/2785 fixes this and resolves this issue.

closed 2020-06-24 by bafonins