TheThingsNetwork/lorawan-stack #1574: Validation errors on copy-pasted location values

htdvisser, 2019-11-11:
#### Summary

When I copy-paste a location value into the Latitude/Longitude of the gateway location, the console gives me (client side) validation errors.

#### Steps to Reproduce

1. Go to your gateway
2. Go to **Location**
3. Copy latitude `52.39528339` and longitude `4.87489745`
4. **Save Changes**

#### What do you see now?

[image]

#### What do you want to see instead?

The console should not complain.

kschiffer, 2019-11-21:
I'm having trouble reproducing. Which browser are you using?

@pgalic96 this is likely fixed by changing the input type from `number` to `text`. But it might need a different validation schema then.

htdvisser, 2019-11-21:
This was on Firefox 70 I think, updated issue.

PR #1893 (2020-01-28): Fix validation and proptype errors in location form

#### Summary

Closes #1574

#### Changes

- Cast input type number into floats to avoid proptype errors (and possible other errors?)
- Add `step` to form to fix issues with form input in Firefox

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [x] Testing: The changes are covered with unit tests. The changes are tested manually as well [... cut]
closed 2020-01-28 by pgalic96