TheThingsNetwork/lorawan-stack #1577: Error creating a multicast group

alexbn71, 2019-11-11:
#### Summary
Can't create a multicat group using CLI with the suggested command in the online giude. 

#### Steps to Reproduce
The command is:
$ ttn-lw-cli end-devices create app1 mc1 \
  --frequency-plan-id EU_863_870 \
  --lorawan-version 1.0.3 \
  --lorawan-phy-version 1.0.3-b \
  --session.dev-addr 00E4304D \
  --session.keys.app-s-key.key A0CAD5A30036DBE03096EB67CA975BAA \
  --session.keys.nwk-s-key.key B7F3E161BC9D4388E6C788A0C547F255 \
  --multicast

two small inaccuracies here: 
1) 1.0.3-b isn't a valid phy (this value isn't accepted)
2) session.keys.nwk-s-key.key is now deprecated (this cause a warning message)

correcting the above:
$ ttn-lw-cli end-devices create app1 mc1 \
  --frequency-plan-id EU_863_870 \
  --lorawan-version 1.0.2 \
  --lorawan-phy-version 1.0.2-b \
  --session.dev-addr 00E4304D \
  --session.keys.app-s-key.key A0CAD5A30036DBE03096EB67CA975BAA \
  --session.keys.f_nwk_s_int_key.key B7F3E161BC9D4388E6C788A0C547F255 \
  --multicast

#### What do you see now?
ERROR Could not create end device, rolling back... error=error:pkg/networkserver:class_a_multicast (multicast device in class A mode)
error:pkg/networkserver:class_a_multicast (multicast device in class A mode)

#### What do you want to see instead?
The correct multicast group created.

#### Environment
v3.2.3

htdvisser, 2019-11-11:
So we have multiple things here:

1. There is no `1.0.3-b` PHY, so https://github.com/TheThingsNetwork/lorawan-stack/blob/master/doc/content/guides/class-c-multicast/_index.md needs to be fixed (2 occurrences)
2. The CLI should try not to complain about `nwk-s-key` vs `f-nwk-s-int-key` if we know that the user is working with a pre-1.1 end device
3. The commands in https://github.com/TheThingsNetwork/lorawan-stack/blob/master/doc/content/guides/class-c-multicast/_index.md need to include `--supports-class-c`

---

@alexbn71 try adding `--supports-class-c` to your `create` command

alexbn71, 2019-11-11:
Unfortunately nothing to do...

ttn-lw-cli end-devices create app1 mc1 \
  --frequency-plan-id EU_863_870 \
  --lorawan-version 1.0.2 \
  --lorawan-phy-version 1.0.2-b \
  --session.dev-addr 00E4304D \
  --session.keys.app-s-key.key A0CAD5A30036DBE03096EB67CA975BAA \
  --session.keys.nwk-s-key.key B7F3E161BC9D4388E6C788A0C547F255 \
  --multicast \
  --supports-class-c

returned the same error: 
ERROR Could not create end device, rolling back... error=error:pkg/networkserver:class_a_multicast (multicast device in class A mode)
error:pkg/networkserver:class_a_multicast (multicast device in class A mode)

rvolosatovs, 2019-11-12:
@alexbn71 if you're able to run the stack from a specific commit, you can try https://github.com/rvolosatovs/lorawan-stack/commit/c75bdce17bef1616d56a908a9d238f1bea26cd2f - I rebased #1583 on top of `v3.2.4`(latest release) in that commit. If you do that, would be great if you could confirm that fixes your issue.
Otherwise you can wait until #1583 lands in a new release.

PR #1583 (2019-11-13): Fix multicast device MAC state construction

#### Summary

Closes #1577 

#### Changes

- Fix multicast device creation

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [x] Testing: The changes are covered with unit tests. The changes are tested manually as well.
- [x] Documentation: Relevant documentation is added or updated.
- [x] Changelog: Significant features, behavior changes, depr [... cut]
closed 2019-11-13 by rvolosatovs