TheThingsNetwork/lorawan-stack #1596: File Descriptor Leak

htdvisser, 2019-11-15:
#### Summary

There seems to be a file descriptor leak somewhere.

#### Steps to Reproduce

1. Just let ttn-lw-stack run for a while

#### What do you see now?

[image]

#### Environment

ttn-lw-stack v3.2.4

#### How do you propose to implement this?

We first need to figure out where this is coming from, so if anyone else is experiencing this, please investigate and comment.

KrishnaIyer, 2020-06-04:
This seems to have been fixed somewhere since `3.2.4.`. Haven't observed this since `3.7.x` (or that's when we have telemetry).

htdvisser, 2020-06-04:
Yeah, let's close it.

closed 2020-06-04 by htdvisser