TheThingsNetwork/lorawan-stack #1601: Console: javascript errors

alexbn71, 2019-11-15:
#### Summary
I have the stack on a public domain (not in localhost) but navigating the TTN Console there are some javascript errors visible in Developer Console of Chrome. The most common is: **http://localhost:1885/api/v3/edtc/formats net::ERR_CONNECTION_REFUSED** .... why localhost?

Another error in dev console is: **https://mydomain.com/api/v3/js/applications/app-lora-prodigy/devices/app-lora-prodigy-mc1?field_mask=root_keys,resets_join_nonces,network_server_address,application_server_address**
.... that return 404

Another issue is that now, with the new FIX in multicast device creation , I created a multicast device succesfully (with CLI) but when I try to update some values in "General Settings" of the Console the "Save" button correctly become enabled but clicking it do nothing.

#### Steps to Reproduce
Navigate applications and devices in Console.

#### Environment
v3.2.5

alexbn71, 2019-11-18:
[image]

here are visible the resources that are strangely retrieved by "localhost"

kschiffer, 2019-11-19:
Hey alexbn71, 
you forgot to add the console configs for the new Device Claiming Server (DCS) and QR Code Generator (QRG). They need to be set up like the other component servers., eg. via config flags:
`--console.ui.edtc.base-url=https://mydomain.com/api/v3` and `--console.ui.qrg.base-url=https://mydomain.com/api/v3`

Concerning the 404: the console will sometimes have to make calls to the device registries (eg. here on the join server) without knowing whether the device is present, so this is not an error per say but an API endpoint rightfully returning 404 for a non-existing device entry.

I will try to recreate your multicast issue and let you know.

alexbn71, 2019-11-19:
Thanks, addding those configurations has resolved the localhost issues.

About 404... if I understand correctly you mean that a multicast device does not have an entry in the JS so it's normal that the resource is unavailable, right?

I confirm that the multicast issue is still there and was unrelated with the above issues.

kschiffer, 2019-11-19:
> About 404... if I understand correctly you mean that a multicast device does not have an entry in the JS so it's normal that the resource is unavailable, right?

Yes. The SDK (which the console uses to make the API calls) is not yet optimized for multicast in the sense that it would avoid JS calls in these cases. I will work on optimizing that soon.

kschiffer, 2019-11-22:
@alexbn71 
We're currently working on rebuilding the device creation and update UI in the console. This will also include proper handling of multicast devices. It will likely take a little more time to implement, but I wouldn't want to touch our current logic when it will be refactored anyway. As such, I will close this issue for now. You can track our progress of this here #579 

Thanks for reporting this!

closed 2019-11-22 by kschiffer