TheThingsNetwork/lorawan-stack #1617: Guide for device maker to pre-provision devices to their join server or global join server

wienke, 2019-11-21:
#### Summary
Guide for device maker to pre-provision devices to their join server or global join server

#### Why do we need this ?
Supporting device makers in the ease of use and the openness of The Things Network global join server.

#### What is already there? What do you see now?
Don't know

htdvisser, 2019-11-21:
Good one. Things to cover here would be:

For the CLI:

- How to configure the CLI to only enable IS and JS
- What fields are necessary (including claim authentication code)
- How to generate a QR code for the device
- Maybe how this works in combination with end device templates?

For the Console:

- How to work with bulk import
- What fields are necessary (including claim authentication code)
- How to generate a QR code for the device
- ...?

Maybe also with the HTTP API?

htdvisser, 2019-11-21:
@adriansmares please see what's already there and what we still need.

KrishnaIyer, 2024-12-24:
This will be handled elsewhere.

closed 2024-12-24 by KrishnaIyer