TheThingsNetwork/lorawan-stack #1626: Support selecting gateways in CLI downlink push

rvolosatovs, 2019-11-21:
#### Summary

CLI does not support selecting gateways on `end-devices downlink push`.
cc @elsalahy 

#### Steps to Reproduce

1. `cli device downlink push --help`

#### What do you see now?

```
<...>
Flags:
      --application-id string            
      --class-b-c.absolute-time string   (YYYY-MM-DDTHH:MM:SSZ)
      --confirmed     
<...>                   
```

#### What do you want to see instead?

```
<...>
Flags:
      --application-id string            
      --class-b-c.absolute-time string   (YYYY-MM-DDTHH:MM:SSZ)
      --class-b-c.gateways # <- this should allow selecting both gateway and the antenna index
      --confirmed     
<...>                   
```
See  https://github.com/TheThingsNetwork/lorawan-stack/blob/3a9223709571fad0cf6ef7bec1ebda06f75f5f9c/api/messages.proto#L204-L215

The flag has to map to https://github.com/TheThingsNetwork/lorawan-stack/blob/3a9223709571fad0cf6ef7bec1ebda06f75f5f9c/api/lorawan.proto#L281-L284 somehow

#### How do you propose to implement this?

Map `[]*ttnpb.GatewayAntennaIdentifiers` to some specific string format in `config` I suppose?

#### Can you do this yourself and submit a Pull Request?

yes, but @htdvisser is probably better suited for the job.

htdvisser, 2019-11-21:
Definitely not going to have time for this in November.

rvolosatovs, 2019-11-22:
This is quite important for @elsalahy and me for testing, setting priority

htdvisser, 2019-11-22:
Ok then it's better if you do it yourself.

rvolosatovs, 2019-11-22:
How about `--class-b-c.gateways my-gateway:1,my-other-gateway` format?
So per gateway-antenna pair: `<gtw-id>[:index]`,
`gtw-id` is mandatory. `index` is optional.
Several pairs are separated by a `,`

htdvisser, 2019-11-22:
We can already start with only supporting `gtw-id`, and then later extend it to also support antenna index.

rvolosatovs, 2019-11-22:
Sure, but this way we may run into issues later once we decide to support the index as well. Approach described in https://github.com/TheThingsNetwork/lorawan-stack/issues/1626#issuecomment-557540425 would work, but if there's a better option, which may be incompatible with string slice later on, I think we should go for that.
Do you have any suggestions?

johanstokking, 2019-11-24:
I think one flag with the gateway ID suffices, assuming index 0. Just keep it simple.

Being able to specify multiple gateways when using the CLI is advanced use that we can support later if there's popular demand. You can already do this by using MQTT or webhooks.

PR #1646 (2019-11-27): Support for `class-b-c.gateways` in CLI

#### Summary

Closes #1626 

#### Changes

- Support for `class-b-c.gateways` in CLI
- Refactor the CLI flag decoding(necessary for :arrow_up:)
- Make `class-b-c.gateways` non-nullable 

#### Notes for Reviewers

In order to deal with the existing issue of new fields breaking JSON compatibility and polluting the output for our users I added https://github.com/TheThingsNetwork/lorawan-stack/commit/5ca539771c28fc2e0f4e36278f2df25e9bb46a90 @htdvisser if you agree with this, I will squash it as intended.

I suggest reviewing this commit-by-commit.

#### Che [... cut]
closed 2019-11-27 by rvolosatovs