TheThingsNetwork/lorawan-stack #1631: Add Kerlink Wirnet connection guide

rvolosatovs, 2019-11-22:
#### Summary

Document connecting Kerlink Wirnet

#### Why do we need this?

To get people to use our network

#### What is already there? What do you see now?

Nothing

#### What is missing? What do you want to see?

A clear guide on how to connect Kerlink Wirnet to a specific v3 stack instance

#### How do you propose to implement this?

- Create a script, which would generate DOTA configured to connect to specified host and use specified FP, which users can install on their gateways. E.g. `generate-dota.sh <my stack instance> <my FP>`
- Document how to use the script to configure the gateway
- Document how to update the gateway to latest firmware
- Provide the latest firmware - I suppose that goes to https://github.com/TheThingsNetwork/kerlink-station-firmware ? (cc @htdvisser) - if so, let's make that repo look a bit nicer?
   1. Let's use releases page for actual packages.
   2. Let's add the script, which would generate the required DOTA there by using the latest firmware release in the repo.

#### Can you do this yourself and submit a Pull Request?

yes

rvolosatovs, 2019-11-22:
Blocked by https://github.com/TheThingsNetwork/lorawan-stack/pull/1619

Let's use the format there as an example

rvolosatovs, 2019-11-22:
cc @johanstokking

htdvisser, 2019-11-22:
@rvolosatovs you're now maintainer on https://github.com/TheThingsNetwork/kerlink-station-firmware

I don't think we should provide Kerlink firmware or packages like the Kerlink Packet Forwarder or Semtech Packet Forwarder, but we can definitely improve the repo.

htdvisser, 2019-11-22:
Assigning @johanstokking and myself for the discussion

rvolosatovs, 2019-11-22:
> I don't think we should provide Kerlink firmware or packages like the Kerlink Packet Forwarder or Semtech Packet Forwarder, but we can definitely improve the repo.

Why not? We already do that in https://github.com/TheThingsNetwork/kerlink-station-firmware, right?
All I propose is to update the versions and do it in a more systematic way.

johanstokking, 2019-11-23:
I think we should have sane defaults, including binaries, and a script to build them and set host names. Also, it would be nice to have pre-built packages for common deployments (PCN regions, CH regions).

rvolosatovs, 2019-11-28:
Blocked until I manage to actually get class B to work

htdvisser, 2019-11-29:
Why would we wait for that? If it can receive uplink and send downlink, we can already document how to make it work for most Class A and Class C use cases.

johanstokking, 2019-12-02:
This should not be blocked by Class B indeed.

Please keep The Things Kickstarter Gateway structure as reference; prerequisites, registration, configuration, troubleshooting.

johanstokking, 2019-12-02:
References #1140

rvolosatovs, 2019-12-03:
The reason why I marked is as "blocked" on class B is because a possible cause for class B not working is a misconfiguration and we don't want our users to have misconfigured gateways

htdvisser, 2019-12-03:
This is still a priority

rvolosatovs, 2019-12-03:
Oh, I thought that was the "blocked" label I was removing, but looks like @johanstokking already removed it

PR #1716 (2019-12-23): Document connecting Kerlink Wirnet Station to TTS

#### Summary

Closes #1631 

#### Changes

- Document connecting Kerlink Wirnet Station to TTS

#### Notes for Reviewers

~Leaving this as draft until https://github.com/TheThingsNetwork/kerlink-station-firmware/pull/7 is merged~

Do we need to document this in the `CHANGELOG`? I guess not, but @htdvisser may have a different opinion.

Based this on existing TTKG guide. ~`Registration` section is just copy-paste~

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes ar [... cut]
closed 2019-12-23 by rvolosatovs