TheThingsNetwork/lorawan-stack #1648: Add user management for admins to the console

kschiffer, 2019-11-27:
#### Summary
Admin users should be able to manage other users via the console

#### Why do we need this?
To provide user management via a web interface (currently only possible via CLI)

#### What is already there? What do you see now?
Nothing of that sort.

#### What is missing? What do you want to see?
An admin area inside the console, where admins can configure users of the deployment. This includes:
- inviting a user
- listing users (API still needed)
- approving/suspending users

#### Environment
Console

#### How do you propose to implement this?
In the user dropdown (top right corner), we can add an "Admin Settings" section, which is only visible when the user has the admin flag. The user management could be a simple view, which lists all users (with search), all unapproved users and a button to invite users via email.
I'll do some quick wireframes for a POC.

#### Can you do this yourself and submit a Pull Request?
Yes.

kschiffer, 2019-12-02:
I'm looking into this now and I'm wondering whether this should be made part of the new Account App rather (#1422) since the management of users appears to me like a superordinate concern.

I believe it could be beneficial to group such overall admin concerns into the account app to also shape it as more of a stack/deployment configuration tool while keeping the console scoped to network-related management. We can make this accessible in the console then by adding links to the account app (preferably in the user dropdown).

Thoughts @johanstokking @htdvisser ?

johanstokking, 2019-12-02:
My take is twofold; one is that we don't have time to wait for #1422 to get basic user management, and two is that I think that what the new account app is going to be once we defined it, is mostly about the user itself, and not about other users.

htdvisser, 2019-12-02:
I agree with @johanstokking's comment. The account app will be for managing "your own user account", the console is for managing the network, and when you're an admin that includes the users of your network.

johanstokking, 2019-12-02:
Pushing this to high to avoid a CLI-only user experience in the initial onboarding phase for deployments.

kschiffer, 2019-12-05:
I would like to add a small table of unapproved users to the user management view in the console. Otherwise, the user would have to enter the user id manually.
Is it worth extending the RPCs for this? Right now we cannot filter the list RPC. Likewise, it would improve the UX a lot if one would be able to search the list of users.

htdvisser, 2019-12-05:
We have the search RPCs for that. We already planned on adding a `state` filter to those.

PR #1706 (2019-12-12): Add user management for admins to the console
#### Summary
Closes #1648 
References: #1086 

[image]

[image]

#### Changes
- Add user service to the JS SDK
- Add routing for admin settings
- Add views and containers for listing and editing users
- Extend feature checks for admin settings
- Extend redux primitives for listing, getting and updating users
- Minor related fixes such as PropTypes fixes

#### Notes for Reviewers
PR got quite big, but a lot of it is just plumbing and extending the redux primitives.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelat [... cut]
closed 2019-12-12 by kschiffer