TheThingsNetwork/lorawan-stack #1654: Add Searching functionality in the Console

vsbandaru, 2019-11-27:
#### Summary
Search functionality in Console by DevEUI, DevID or Name, more fields could be added. 

#### Why do we need this?
Looking up device status to debug/monitor gets harder as the number of devices deployed reaches hundreds or thousands. Most organizations would follow a naming convention for the Device ID/Name and having the Search feature eases the management. 

#### What is already there? What do you see now?
No Search functionality available currently.

#### What is missing? What do you want to see?
A Search UI on the Applications page with the following fields

- Device ID
- Device Name
- Dev EUI
- and more if suggested by other users.

#### Environment
Console

#### How do you propose to implement this?
Implement a Search Header Section on **Applications** page.

#### Can you do this yourself and submit a Pull Request?
No

kschiffer, 2019-11-28:
Related: https://github.com/TheThingsNetwork/lorawan-stack/issues/73

htdvisser, 2019-11-28:
API definition for EndDeviceRegistrySearch: https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.3.1/api/search_services.proto#L84-L115

That API allows you to (substring) search by

- ID
- Friendly Name
- Description
- Attributes
- EUI
- DevAddr (*)

A simple version of this API can be implemented on the Identity Server. This implementation would be able to search by all above fields, except DevAddr (the DevAddr is owned by the Network Server, not the Identity Server). I can work on that.

This would be simpler than resolving #73, because we can already scope the search to the Application ID that we know the caller has rights on.

Let's separate the issues (I'll remove EndDevice search form #73) and focus on this one (#1654) first.

kschiffer, 2020-01-08:
Closing this, since it intersects with #1803. Please continue discussion there if need be.
The issue has a PR and should be resolved soon.

PR #1803 (2020-01-24): Entity search by ID in console
#### Summary
Closes #1802 
Closes #1654 

[image]

#### Changes
- Extend SDK services to include search endpoints
- Update redux logics to enable search
- Update containers to include search

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [x] Testing: The changes are covered with unit tests. The changes are tested manually as well.
- [x] Documentation: Rel [... cut]
closed 2020-01-08 by kschiffer