TheThingsNetwork/lorawan-stack #1659: AS flaky test

johanstokking, 2019-11-28:
#### Summary

AS flaky test

#### Steps to Reproduce

Run CI and flip a coin

#### What do you see now?

```
  WARN No cookie hash key configured, generated a random one hash_key=01AE55CC09953255FF9840534C9D384A311D8E9BFC586A04DCF2A65667BDDA31DD2D192C6F03596060168F7F5019C768702676EDD3820BFDC62D5FE0B363D896 namespace=web test_name=TestApplicationServer
  WARN No cookie block key configured, generated a random one block_key=FD5B7C6F06E4835B27CF5D13C9602E2940E4864EE5FABA33E6E892821D4C1622 namespace=web test_name=TestApplicationServer
  WARN No static assets found in any search path namespace=web search_paths=[] test_name=TestApplicationServer
  WARN No rights TTL configured                 test_name=TestApplicationServer
  WARN No cluster key configured, generated a random one key=e42dbad1c83222ca4f41eceda23e0d064524db63e77cf750efc78bdcdd934eae test_name=TestApplicationServer
  INFO Listening for connections                address=:9184 namespace=grpc protocol=gRPC test_name=TestApplicationServer
  INFO Listening for connections                address=:8099 namespace=web protocol=Web test_name=TestApplicationServer
  INFO Linked                                   application_uid=foo-app namespace=applicationserver network_server=ns test_name=TestApplicationServer
  INFO Subscribed                               application_uid=foo-app grpc_method=Subscribe grpc_service=ttn.lorawan.v3.AppAs namespace=applicationserver/io/grpc remote_addr=pipe request_id=01DTRSZ4VQRHGVC86X6P6NXW7E test_name=TestApplicationServer
  INFO Unsubscribed                             application_uid=foo-app grpc_method=Subscribe grpc_service=ttn.lorawan.v3.AppAs namespace=applicationserver/io/grpc remote_addr=pipe request_id=01DTRSZ4VQRHGVC86X6P6NXW7E test_name=TestApplicationServer
  WARN Finished streaming call                  duration=1.024010206s error=context deadline exceeded grpc_method=Subscribe grpc_service=ttn.lorawan.v3.AppAs namespace=grpc request_id=01DTRSZ4VQRHGVC86X6P6NXW7E test_name=TestApplicationServer
  INFO Unlinked                                 application_uid=foo-app namespace=applicationserver network_server=ns test_name=TestApplicationServer
  WARN transport: http2Server.HandleStreams failed to read frame: io: read/write on closed pipe namespace=grpc test_name=TestApplicationServer
--- FAIL: TestApplicationServer (2.67s)
    --- FAIL: TestApplicationServer/Authenticate/grpc (1.02s)
        --- FAIL: TestApplicationServer/Authenticate/grpc/ValidIDAndKey (1.02s)
            applicationserver_test.go:833: Expected deadline exceeded with valid auth, but have rpc error: code = Unknown desc = context deadline exceeded
FAIL
FAIL	go.thethings.network/lorawan-stack/pkg/applicationserver	2.733s
```

#### What do you want to see instead?

Test pass

#### Environment

Travis

#### How do you propose to implement this?

Increase time-out at that specific place

#### Can you do this yourself and submit a Pull Request?

Can review

neoaggelos, 2020-02-28:
And another one https://github.com/TheThingsNetwork/lorawan-stack/pull/2036/checks?check_run_id=475348591:

```
panic: runtime error: invalid memory address or nil pointer dereference [recovered]
	panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x14c2bb4]

goroutine 59 [running]:
testing.tRunner.func1(0xc000124600)
	/opt/hostedtoolcache/go/1.13.8/x64/src/testing/testing.go:874 +0x3a3
panic(0x163f1c0, 0x282ba30)
	/opt/hostedtoolcache/go/1.13.8/x64/src/runtime/panic.go:679 +0x1b2
go.thethings.network/lorawan-stack/pkg/applicationserver/io/pubsub/provider/nats_test.TestOpenConnection.func1.1(0xc000124600)
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/applicationserver/io/pubsub/provider/nats/provider_test.go:167 +0x334
testing.tRunner(0xc000124600, 0xc00047dad0)
	/opt/hostedtoolcache/go/1.13.8/x64/src/testing/testing.go:909 +0xc9
created by testing.(*T).Run
	/opt/hostedtoolcache/go/1.13.8/x64/src/testing/testing.go:960 +0x350
FAIL	go.thethings.network/lorawan-stack/pkg/applicationserver/io/pubsub/provider/nats	2.095s
```

neoaggelos, 2020-04-11:
Got this one again:

```
--- FAIL: TestApplicationServer (3.68s)
    --- FAIL: TestApplicationServer/Authenticate/grpc (2.05s)
        --- FAIL: TestApplicationServer/Authenticate/grpc/ValidIDAndKey (2.05s)
            applicationserver_test.go:834: Expected deadline exceeded with valid auth, but have error:unknown:unknown (context deadline exceeded)
FAIL
```

Looks very similar to the one we got in GS

johanstokking, 2020-04-30:
@htdvisser saw this one today several times as well

adriansmares, 2020-05-01:
As of #2469 the timeout for connection tests (including gRPC) has been increased. Please let me know if you keep encountering the following issue:
```
--- FAIL: TestApplicationServer (3.68s)
    --- FAIL: TestApplicationServer/Authenticate/grpc (2.05s)
        --- FAIL: TestApplicationServer/Authenticate/grpc/ValidIDAndKey (2.05s)
            applicationserver_test.go:834: Expected deadline exceeded with valid auth, but have error:unknown:unknown (context deadline exceeded)
FAIL
```

Note that since the PR was merged to `master` it may take a while until it is merged to `develop`.

closed 2020-06-09 by htdvisser