TheThingsNetwork/lorawan-stack #1661: Console does not show warnings

htdvisser, 2019-11-28:
#### Summary

The console doesn't show warnings sent by APIs

#### Steps to Reproduce

1. `make dev.databases.sql`
2. `UPDATE users SET primary_email_address_validated_at = NULL;`
3. `go run ./cmd/ttn-lw-stack start --is.user-registration.contact-info-validation.required`
4. Open the console

#### What do you see now?

[image]

The requests to http://localhost:1885/api/v3/auth_info and http://localhost:1885/api/v3/users/htdvisser return `X-Warning: Restricted rights until email address validated` in the headers. Other API requests return 403.

#### What do you want to see instead?

1. The Console must display these warnings (they can also be about other problems)
2. The Console should probably also show something other than "Fetching data..." if API requests return errors.

#### Environment

v3.3.1

kschiffer, 2019-12-09:
There's some related discussion in here as well:
https://github.com/TheThingsNetwork/lorawan-stack/pull/1151

I need to understand this a little better. In what other situations do we add a warning header to the API responses? From what I see now, I think it makes most sense to forward the warning to a toast notification in the console (small notification in the bottom right corner), but this would be problematic if we have other types of warnings that would be displayed with every API response (as it would flood the toast message queue). We'd then have to add logic to make the warnings dismissable.

htdvisser, 2019-12-10:
To see what warnings we currently have, search for `warning.Add(`.
I don't know what we might add in the future.

htdvisser, 2019-12-10:
Let's start by logging them to the console.

johanstokking, 2019-12-10:
Can't we have a scrolling warning console in the footer, much like what's shown in developer tools? With an incrementing counter? And on each entry a timestamp or something

PR #1733 (2020-01-03): Show warning headers in the console
#### Summary
Closes #1661 

This PR will make warning header (`X-Warning: …`) display in the console frontend as toast notifications. It also introduces some related refactoring to the login flow.

[image]

#### Changes
- Allow restricted users to view at least their own user info (IS)
- Add a simple event system to the JS SDK
- Emit a warning event when the backend response contains a warning header
- Add oauth info service to JS SDK (so we can replace the plain api request with the SDK equivalient)
- Add automatic adding of `STATE_REQUESTED` state to the J [... cut]
closed 2020-01-03 by kschiffer