TheThingsNetwork/lorawan-stack #1663: Add endpoint to retrieve available LoRaWAN versions

kschiffer, 2019-11-29:
#### Summary
The console needs to know the available LoRaWAN versions, to suggest them during end device creation.

#### Why do we need this?
We currently have this information hard coded into the console.

#### What is already there? What do you see now?

- Device creation/update view
- Fetch Select component to show options that were retrieved from the backend

#### What is missing? What do you want to see?
- An endpoint to retrieve the list of available LoRaWAN versions
- Wiring it into the frontend

#### Environment

v3.3.2

#### Can you do this yourself and submit a Pull Request?

I can do the frontend part

@htdvisser please reassign if appropriate

htdvisser, 2019-11-29:
This is something for the JS SDK.

Should come from https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/lorawan.proto#L65-L89

htdvisser, 2019-11-29:
It's probably going to be sufficient to add these to [pkg/ttnpb/i18n.go](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/ttnpb/i18n.go), but the custom stringers for [MACVersion](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/lorawan.proto#L65-L89) and [PHYVersion](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/ttnpb/lorawan.go#L725-L744) might complicate this a bit.

htdvisser, 2020-08-24:
This doesn't belong in an API endpoint, since it can be generated from the protos.

closed 2020-08-24 by htdvisser