TheThingsNetwork/lorawan-stack #1664: Make console navigation bar responsive

kschiffer, 2019-11-29:
#### Summary

As part of making the console responsive (#1172), we need to make the navigation bar responsive.

#### Why do we need this?

To provide proper experience on mobile devices.

#### What is already there? What do you see now?

Navigation bar

#### What is missing? What do you want to see?

The navigation bar collapsing properly on smaller screens. The navigation entries need to be accessible from a dropdown menu then.

#### Environment

Console

#### How do you propose to implement this?
CSS magic with some JS wizardry

#### Can you do this yourself and submit a Pull Request?

Yes

bafonins, 2020-06-16:
Shouldnt this be closed?

kschiffer, 2020-06-19:
Indeed.

closed 2020-06-19 by kschiffer