TheThingsNetwork/lorawan-stack #1685: Document migrating devices from V2 to V3

johanstokking, 2019-12-03:
#### Summary

Document migrating devices from V2 to V3

Needs https://github.com/TheThingsNetwork/ttn/issues/764

#### Why do we need this ?

To get started quickly

#### What is already there? What do you see now?

- Export devices in V3 JSON format with V2 `ttnctl` (https://github.com/TheThingsNetwork/ttn/issues/764)
- CLI documentation and getting started

#### What is missing? What do you want to see?

How to migrate devices from V2 to V3 using CLI

#### How do you propose to document this?

Add a guide for upgrading from V2 to V3, potentially with #1684 as well, if we put that in the guide

#### Can you do this yourself and submit a Pull Request?

Can review

neoaggelos, 2020-04-27:
Do we need anything like that for more entities? For example:

- Applications: payload formatters (or maybe include them on the exported devices?)
- Gateways: brand information, frequency plan, locations

The v3 model is quite different, but there is some use in being able to migrate "automatically" some of that information. There are also actions that can't be automated (re-configuring gateways, some integrations) or are hard to automate (e.g. application payload formatters, because they need the application to be linked)

htdvisser, 2020-04-28:
Applications: I'd say no. In v2 the only thing an application really contains is indeed the formatters, and we can't easily migrate those. In order to do so, we would have to parse them, and try to rewrite them to the new v3 data structures, which is a lot of work, and is not guaranteed to be correct.

Integrations: Definitely no. Those can only be managed with the v2 console, not with API, so if we'd want this, we'd have to build it into the v2 console. In addition, the integrations work a bit differently in v3, so it might become confusing if we only partly migrate them, and that they start behaving differently.

Gateways: Maybe. Users already have to generate v3 API keys and reconfigure their gateways. Then it isn't a lot of work to click the register button in v3 first.

neoaggelos, 2020-04-29:
I'm having trouble telling apart OTAA from ABP devices from v2 with ttnctl. What I've seen is that all devices (both OTAA and ABP) have `ActivationConstraints` equal to `Local`. The console seems to be  telling them apart by looking up the usages for devaddr prefixes in https://console.thethingsnetwork.org/api/handlers.

`ttnctl discover broker --metadata` is the closest I can get, but it does not include the usages of the prefixes.

Can we retrieve this info from the handler itself?

PR #2467 (2020-05-11): Document migrating end devices from v2

#### Summary

Closes #1684
Closes #1685 

#### Changes

- Write migration guide.
- This is the documentation part. It depends on https://github.com/TheThingsNetwork/ttn/pull/776, a new ttnctl binary release, and a new v3 release with https://github.com/TheThingsNetwork/lorawan-stack/pull/2457

#### Notes for Reviewers

@benolayinka There are probably quite a few inconsistencies in writing style, probably some missing information
@htdvisser @johanstokking Scope and correctness, especially in the Major Changes page.
@laurensslats Scope, probably coordinate [... cut]
closed 2020-05-11 by neoaggelos