TheThingsNetwork/lorawan-stack #1692: Console Username error - display

rish-c, 2019-12-04:
#### Summary

When creating an invalid username, the console shows an error which could be improved with a simple message.

...

#### Steps to Reproduce

Usernames such as `abc_123` plus other similar invalid usernames that are not allowed

#### What do you see now?

Screenshot
[image]

#### What do you want to see instead?

Display the format of a valid username

#### Can you do this yourself and submit a Pull Request?

@kschiffer @bafonins

kschiffer, 2019-12-05:
Shouldn't the backend take care of providing more helpful error messages? We could add some mapping of error messages, but I think this should be a last resort only.
This one could already be improved greatly by a more expressive error message than `invalid identifiers`, e.g. `User ID contained invalid characters, please use {allowed characters} only`

htdvisser, 2020-08-24:
The frontend already has the regex for valid IDs, so it can add that on the field.

johanstokking, 2020-08-24:
Consider other similar fields too

bafonins, 2020-10-05:
Already implemented.

[image]

closed 2020-10-05 by bafonins