TheThingsNetwork/lorawan-stack #1694: Subscribe to Multiple Applications in Single MQTT Connection

htdvisser, 2019-12-05:
#### Summary

It would be really cool if we could subscribe to multiple applications in a single MQTT connection.

#### Why do we need this?

Simplification

#### What is already there? What do you see now?

MQTT sessions are per-application

#### What is missing? What do you want to see?

It may not be necessary to have this restriction.

#### How do you propose to implement this?

- On `CONNECT` only verify that the auth (API key) is valid
- On `SUBSCRIBE` verify that the auth has read access to the application in the topic filter
	- This requires the `SUBSCRIBE` to explicitly contain the application ID, so no wildcards.
- On `PUBLISH` verify that the auth has write access to the application in the topic

#### Can you do this yourself and submit a Pull Request?

I can investigate and make a proposal. Definitely something for the longer term though.

GMouaad, 2020-01-10:
Is it possible in the future to enable the Stack Owner (the admin) to subscribe to all topics. Or is there anything that speaks against it? I can imagine that an argument would be the data ownership.

htdvisser, 2022-07-07:
Removing my assignment. If there are more users that would like to see this functionality, please react with a 👍 vote on the issue, so that we can track demand from the community.

KrishnaIyer, 2025-04-30:
This is a pretty cool requirement but MQTT is not the most scalable protocol and this isn't something that we should spend time on.