TheThingsNetwork/lorawan-stack #1700: upgrade to V3.3.2 fails

ktsolakas, 2019-12-09:
#### Summary
We applied the V3.3.2 cloudformation template to a 3.2.6 TTI instance and failed to start
#### Steps to Reproduce

1. Apply the 3.3.2 template over a 3.2.6 instance
2. Log into the backend
3. View logs
4. Stack does not start

#### What do you see now?
Dec 09 03:42:04 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: lorawan-stack.service: main process exited, code=exited, status=255/n/a
Dec 09 03:42:04 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: Unit lorawan-stack.service entered failed state.
Dec 09 03:42:04 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: lorawan-stack.service failed.
Dec 09 03:42:07 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: lorawan-stack.service holdoff time over, scheduling restart.
Dec 09 03:42:07 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: Started The Things Enterprise Stack.
Dec 09 03:42:07 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: Starting The Things Enterprise Stack...
Dec 09 03:42:07 ip-10-0-0-163.ap-southeast-2.compute.internal tti-lw-stack[4831]: proto: illegal wireType 6
Dec 09 03:42:07 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: lorawan-stack.service: main process exited, code=exited, status=255/n/a
Dec 09 03:42:07 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: Unit lorawan-stack.service entered failed state.
Dec 09 03:42:07 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: lorawan-stack.service failed.
Dec 09 03:42:10 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: lorawan-stack.service holdoff time over, scheduling restart.
Dec 09 03:42:10 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: Started The Things Enterprise Stack.
Dec 09 03:42:10 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: Starting The Things Enterprise Stack...
Dec 09 03:42:11 ip-10-0-0-163.ap-southeast-2.compute.internal tti-lw-stack[4840]: proto: illegal wireType 6
Dec 09 03:42:11 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: lorawan-stack.service: main process exited, code=exited, status=255/n/a
Dec 09 03:42:11 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: Unit lorawan-stack.service entered failed state.
Dec 09 03:42:11 ip-10-0-0-163.ap-southeast-2.compute.internal systemd[1]: lorawan-stack.service failed.
^C
[ec2-user@ip-10-0-0-163 lorawan-stack]$ sudo systemctl stop lorawan-stack.service

[ec2-user@ip-10-0-0-163 lorawan-stack]$ sudo ./tti-lw-stack start -c config.yml
proto: illegal wireType 6
#### What do you want to see instead?

The stack starting

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

johanstokking, 2019-12-09:
@ktsolakas can you provide debug logs?

Looks like garbage; wire type 6 doesn't exist (see https://developers.google.com/protocol-buffers/docs/encoding).

It would be very helpful to figure out with debug logs which message is being loaded.

ktsolakas, 2019-12-09:
[issue1700.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/3938548/issue1700.txt)
@johanstokking I have attached verbose journactl output. Debug logs are enabled in config.yml

johanstokking, 2019-12-09:
`journalctl` verbose is not needed and makes it harder to read. I don’t see debug logs in this file, that might be the display format, wrong config or indeed a really fast failure.

johanstokking, 2019-12-09:
Closed offline; no resolution details applicable here

closed 2019-12-09 by johanstokking