TheThingsNetwork/lorawan-stack #1701: Downlink timming issue

ktsolakas, 2019-12-09:
#### Summary
After sending a downlink to a device the Gateway logs are showing errors delivering the downlink

#### Steps to Reproduce
1. Sent a downlink to the device
ttn-lw-cli applications downlink replace --application-id meshed-mcf88 --device-id lw130-f016 --frm-payload 0400FFFFFFFF00000000 --f-port 1

#### What do you see now?
Gateway Logs show:
##### 2019-12-05 02:54:13 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 1
# CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
# RF packets forwarded: 1 (23 bytes)
# PUSH_DATA datagrams sent: 2 (324 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 2 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:41, rej:0)
# TX rejected (collision beacon): 0.00% (req:41, rej:0)
# TX rejected (too late): 12.20% (req:41, rej:5)
# TX rejected (too early): 14.63% (req:41, rej:6)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 1142646005
src/jitqueue.c:449:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# Invalid time reference (age: 1575514453 sec)
# no valid GPS coordinates available yet
##### END #####

JSON up: {"stat":{"time":"2019-12-05 02:54:13 GMT","rxnb":1,"rxok":0,"rxfw":1,"ackr":100.0,"dwnb":0,"txnb":0}}
INFO: [up] PUSH_ACK received in 83 ms
INFO: [down] PULL_RESP received  - token[0:42] :)

JSON down: {"txpk":{"imme":false,"tmst":1166001045,"freq":923.2,"rfch":0,"powe":24,"modu":"LORA","datr":"SF10BW125","codr":"4/5","ipol":true,"size":23,"ncrc":true,"data":"YHA/5wGADwABak0aGz1wG9AJ6flBVYU="}}
src/jitqueue.c:234:jit_enqueue(): ERROR: Packet REJECTED, already too late to send it (current=1165970347, packet=1166001045, type=0)
ERROR: Packet REJECTED (jit error=1)
INFO: [down] PULL_ACK received in 79 ms
INFO: [down] PULL_ACK received in 74 ms

##### 2019-12-05 03:04:14 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 1
# CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
# RF packets forwarded: 1 (126 bytes)
# PUSH_DATA datagrams sent: 2 (460 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 2 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:46, rej:0)
# TX rejected (collision beacon): 0.00% (req:46, rej:0)
# TX rejected (too late): 13.04% (req:46, rej:6)
# TX rejected (too early): 13.04% (req:46, rej:6)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 1742676650
src/jitqueue.c:449:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# Invalid time reference (age: 1575515054 sec)
# no valid GPS coordinates available yet
##### END #####

JSON up: {"stat":{"time":"2019-12-05 03:04:14 GMT","rxnb":1,"rxok":0,"rxfw":1,"ackr":100.0,"dwnb":0,"txnb":0}}
INFO: [up] PUSH_ACK received in 91 ms
INFO: [down] PULL_ACK received in 77 ms
INFO: [down] PULL_RESP received  - token[0:47] :)

JSON down: {"txpk":{"imme":false,"tmst":3762079244,"freq":923.2,"rfch":0,"powe":24,"modu":"LORA","datr":"SF10BW125","codr":"4/5","ipol":true,"size":23,"ncrc":true,"data":"YHA/5wGAFAABTQhzF4HsWS92zBJFDgA="}}
src/jitqueue.c:252:jit_enqueue(): ERROR: Packet REJECTED, timestamp seems wrong, too much in advance (current=1779165021, packet=3762079244, type=0)
ERROR: Packet REJECTED (jit error=2)
INFO: [down] PULL_ACK received in 318 ms

#### What do you want to see instead?
Downlink succesfully sent to the device

rvolosatovs, 2019-12-09:
Thanks for reporting the issue.
Which version are you running?

ktsolakas, 2019-12-09:
V 3.2.2

johanstokking, 2019-12-09:
@ktsolakas can you provide a bit more reproduction steps? I.e. is this class C?

ktsolakas, 2019-12-10:
@johanstokking Yes the device is Class C but i have seen the same issue with Class A downlinks as well. Also i have tried 2 Class C devices with the same result and tried 2 different gateways using different packet forwarders (legacy and mp). I will send logs via e-mail to you and @rvolosatovs

johanstokking, 2019-12-10:
@ktsolakas thanks.

Timing is indeed off. Looks like the following situation;

1. connect gateway, potentially receive uplink message
2. restart packet forwarder (or entire gateway)
3. gateway is still connected according to GS (because our expiry is 5 minutes – way too long)
4. send downlink message

Can you try two things:

1. restart TTS after a packet forwarder or gateway restart. This resets the timing information in GS. There should be a log and event that the gateway connects.
2. ensure you receive an uplink message from the gateway before scheduling a downlink message. This should also reset the timing information in GS.

If that fixes the situation, you can already tweak a few things in `config.yml`:

```yaml
gs:
  udp:
    connection-expires: '40s'
    downlink-path-expires: '15s'
    addr-change-block: '1m'
```

ktsolakas, 2019-12-10:
@johanstokking I followed the steps you indicated and sent a bunch of downlinks I only got 1 Packet REJECTED. Sending logs.

johanstokking, 2019-12-10:
Thanks. Your latency is quite variable:

[image]

Is this a cellular backhaul?

Also, looking at the scheduled times, this is all normal:

Start | Now | Delta
-- | -- | --
562494137732 | 562824132327 | 329994595
563519105361 | 568286188000 | 4767082639
586782230324 | 587112224423 | 329994099
587791758696 | 592574459000 | 4782700304
608511901389 | 608841894933 | 329993544
609522259868 | 614304379000 | 4782119132
626159409778 | 626489404320 | 329994542
627173560073 | 631951668000 | 4778107927
638202798060 | 638532790950 | 329992890
639213041615 | 643994732000 | 4781690385
662689718327 | 663019713665 | 329995338
663699386082 | 668482148000 | 4782761918
669068206068 | 669398200757 | 329994689
671910288905 | 676689859000 | 4779570095
676088305171 | 680838844000 | 4750538829
737370486831 | 737700481263 | 329994432

These deltas are in nanoseconds. The ones that are near 5s are join-accepts or class A downlink with RX1 delay of 5 seconds, and the ones around 330ms are class C downlinks.

Coming back to the chart above; that 330ms is tricky if you observe those latency spikes. They're just not hitting 330ms. Note that the packet is 8,9 ms late, which is basically almost on time.

What happens if you set:

```bash
$ ttn-lw-cli gtw set eui-00800000a0004c9f --schedule-downlink-late=false
```

This writes the downlink immediately to the gateway. I don't think it matters, but please do a test.

The 330 ms comes from https://github.com/TheThingsNetwork/lorawan-stack/commit/9d53772d92798b32954b3bc50287512df5cc99e1#diff-77b50648cfc46b1bf4407d1aa1516312R46. We may need to increase that value, to give more time to higher latency gateways.

ktsolakas, 2019-12-11:
@johanstokking yes it is a cellular back-haul. I applied the timeouts and set the --schedule-downlink-late=false to the gateway and some initial test were successful. I will continue to monitor for any mistimed packets.

johanstokking, 2019-12-11:
Two things need to be done to close this issue;

- [ ] Increase the `ScheduleTimeLong` to 500 ms
- [ ] Change the default UDP timing to:

```yaml
gs:
  udp:
    connection-expires: '40s'
    downlink-path-expires: '15s'
    addr-change-block: '1m'
```

ktsolakas, 2019-12-16:
@johanstokking I have applied the gs: udp settings and i am seeing:
src/jitqueue.c:252:jit_enqueue(): ERROR: Packet REJECTED, timestamp seems wrong, too much in advance
I will send you the relevant logs via e-mail!
Thnx

johanstokking, 2019-12-18:
There are a few timing related issues here: https://github.com/TheThingsNetwork/lorawan-stack/issues/1709

We're currently working on fixing those. I think they're related to the problem you're seeing.

closed 2019-12-30 by johanstokking

ktsolakas, 2019-12-31:
@johanstokking  I have upgraded to 3.4.1 and still seeing timing issues. Attaching the lora-pkt-fwd log
[lora-pkt-fwd-31_12_2019.log](https://github.com/TheThingsNetwork/lorawan-stack/files/4011927/lora-pkt-fwd-31_12_2019.log)

johanstokking, 2019-12-31:
Yes that's completely off. Can you send me the corresponding server logs via a private channel? We need to have both server and gateway logs of the same events to figure out what's going on.

ktsolakas, 2020-01-02:
Sending now via mail

PR #1787 (2019-12-30): Fixes for various downlink timing issues

#### Summary

Fixes #1351 
Fixes #1701
Fixes #1707 
Fixes #1708
Fixes #1709

#### Changes

- Fix class A downlink scheduling when an uplink message has been received between the triggering uplink message.
- Reduce the downlink path expiry window to 15 seconds, i.e. typically missing three `PULL_DATA` frames.
- Reduce the connection expiry window to 40 seconds.
- Reduce default UDP address block time from 5 minutes to 1 minute. This allows for faster reconnecting if the gateway changes IP address. The downlink path and connection now expire before the U [... cut]