TheThingsNetwork/lorawan-stack #1703: RPCs for Purging Entities as Admin

htdvisser, 2019-12-09:
#### Summary

Since our soft-delete prevents entities from being recreated, it would be very useful to have (admin-only) RPCs that completely purge an entity and all associated DB models.

#### Why do we need this?

Automatic GC/purge (#604) isn't happening yet, and users still want to be able to recreate entities.

#### What is already there? What do you see now?

See #604

#### What is missing? What do you want to see?

Until we have that, manual purge.

#### How do you propose to implement this?

Additional RPCs, possibly only for gRPC and CLI to start with.

#### Can you do this yourself and submit a Pull Request?

Yes

virtualguy, 2020-07-21:
@htdvisser I'll upvote this one in favor of #604 as this would be perfect for my use case

pgalic96, 2020-11-02:
@htdvisser I remember you saying that you will comment on this issue with some additional info regarding this, seems clear enough with the linked issues, is there anything to add from your side?

htdvisser, 2020-11-02:
I haven't had time to investigate yet, but I can imagine that it's not as straightforward as it sounds.

From the top of my head:

- You might run into surprises with rights checks, as the caller may not have rights to an entity that doesn't really exist
- You'd have to consider linked models:
  - Permanently deleting a gateway should also delete antenna models, attribute models, contact info models
  - Permanently deleting an application should only be possible if there are really no end devices registered anymore
  - Permanently deleting a user could mean that other entities get "orphaned" because there is no (restorable) owner anymore.
- Probably more things I can't immediately think of

pgalic96, 2020-11-04:
@htdvisser 

>   * Permanently deleting a user could mean that other entities get "orphaned" because there is no (restorable) owner anymore.

In this case it makes most sense to me to require the user to not own any entities before purging. Would this make sense?

htdvisser, 2020-11-04:
There's a separate discussion about this in https://github.com/TheThingsIndustries/lorawan-stack-support/issues/184

pgalic96, 2020-11-10:
> Sounds good to me. Let's do that check on the initial (soft) deletion, because that's when we interact with the user. The purge may be done by an admin, but in the future also by an automated cron job, and we don't want those to fail.

So right now I made it possible to purge entities no matter if they are already soft-deleted or not. In this implementation purge always performs all the checks that soft-delete also does, so for example if application still has devices, the purge will fail (as you mentioned in the comment in this issue). Judging by your comment, should purge be restricted by the rules that apply to soft-delete, either for application-device relation or for user-organization relation, or should it not care about these validations, which would ensure it will always succeed in case of a cron job? We could also control this behavior with a cli flag like `--force`

johanstokking, 2020-11-10:
I'm strongly in favor of having deletion checks on soft-delete. As long as we don't have an option to undo a soft-delete, and purging would fail because of additional checks (i.e. app has devices), we can still get in an invalid state.

pgalic96, 2020-11-10:
Yeah I didn't mean removing the checks for soft-delete, just whether we should enforce the same checks for purge, like the organization rights check that we discussed (or if app contains devices as well). For me it would make sense to have them, since if the checks are not performed we might end up in an invalid state anyways. @htdvisser suggested that we shouldn't perform these checks for purge, as it might cause the automated job to fail. I think having a flag that can bypass these checks (or at least the ones related to organization rights) would do the job, while without the flags the purge would do soft-delete checks but would in fact hard delete all linked entries.

PR #3467 (2020-12-03): Purge entities from db

#### Summary

Closes #1703 

#### Changes

- Add RPC for users, organizations, gateways and applications
- Add cli commands for purging entities

#### Testing

Manual, unit testing

##### Regressions

Should not be any as this is a new functionality

#### Notes for Reviewers

There are no rights checks and rights transfer for organizations and users on purge (it will be part of soft-delete checks per https://github.com/TheThingsNetwork/lorawan-stack/issues/3458 ), as this is an admin-only functionality. 
#### Checklist

- [x] Scope: The referenced issue is addressed, th [... cut]
closed 2020-12-03 by pgalic96