TheThingsNetwork/lorawan-stack #1705: Validate `user_id` in OAuth views

htdvisser, 2019-12-09:
#### Summary

We need to do better frontend validation of the user ID field in OAuth views.

#### Why do we need this?

We're seeing that users are making too many mistakes with user IDs in the OAuth views

#### What is already there? What do you see now?

First of all, the validations are inconsistent. We sometimes see

```js
user_id: Yup.string().required(sharedMessages.validateRequired)
```

while in other places we also see a min/max length.

#### What is missing? What do you want to see?

The user_id should be validated the same way everywhere:

- Required
- Min length 3
- Max length 36
- Regex `^[a-z0-9](?:[-]?[a-z0-9]){2,}$`

> NOTE: maybe out of scope for this issue, but it might be good to check all ID fields in the console as well. All IDs have the same rules.

#### How do you propose to implement this?

Write once, copy-paste many.

pgalic96, 2019-12-12:
@htdvisser What should be the feedback to user if their username doesn't match the regular expression? I can just output the regex value, or show some short info (which is a bit hard to explain in one line).

htdvisser, 2019-12-12:
Commented in PR

PR #1725 (2019-12-17): Improve user id validation in oauth

#### Summary

Closes #1705 

#### Changes

- Add user regex
- Improve user validation and add it to all `oauth` views

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [ ] Testing: The changes are covered with unit tests. The changes are tested manually as well.
- [ ] Documentation: Relevant documentation is added or updated.
- [ ] Changelog: Significa [... cut]
closed 2019-12-17 by pgalic96