TheThingsNetwork/lorawan-stack #1708: Basic Station Integration: Clock misalignment after long periods of inactivity

beitler, 2019-12-09:
#### Summary
lorawan-stack's gatewayserver truncates `xtime` to 32 bit and tracks the rollovers by detecting backward jumps:

https://github.com/TheThingsNetwork/lorawan-stack/blob/dc8fc5ab888f57c20657729b51760eca98452605/pkg/gatewayserver/scheduling/clock.go#L83

This function is called only when there is activity (uplink messages) from a gateway. If the gateway has no activity for more than a rollover epoch (~71 minutes), the RolloverClock.absolute reference will get out of sync because a whole epoch is missed. The consequence is that no downlinks can be scheduled beyond that point until the gateway re-initializes the radio.

#### Steps to Reproduce
0. Connect the gateway, send a few uplinks.
1. Leave the gateway running in a quiet envirnoment for more than 71 minutes.
2. Send an uplink forcing a downlink to be scheduled over the same gateway

#### What do you see now?

The `xtime` field contained in the `dnmsg` will be in the past. The downlink will be rejected. The same will happen to all future downlinks.

#### Environment
Basic Station, TTN public network.

#### How do you propose to implement this?
Ideally, don't downconvert xtime to 32 bit. If this is not an option, inside the rollover tracking function check for multiple rollovers by checking against system time clock monotonic passed between subsequent calls.

johanstokking, 2019-12-09:
Thanks, this helps a lot with reproducing.

> by checking against system time clock monotonic passed between subsequent calls.

This is what we should be doing indeed.

KrishnaIyer, 2019-12-26:
Indeed this is what I should've done for the SessionID. Instead of just using the session ID from the initial connect, the number of rollovers should've been added to it.

PR #1787 (2019-12-30): Fixes for various downlink timing issues

#### Summary

Fixes #1351 
Fixes #1701
Fixes #1707 
Fixes #1708
Fixes #1709

#### Changes

- Fix class A downlink scheduling when an uplink message has been received between the triggering uplink message.
- Reduce the downlink path expiry window to 15 seconds, i.e. typically missing three `PULL_DATA` frames.
- Reduce the connection expiry window to 40 seconds.
- Reduce default UDP address block time from 5 minutes to 1 minute. This allows for faster reconnecting if the gateway changes IP address. The downlink path and connection now expire before the U [... cut]
closed 2019-12-30 by johanstokking