TheThingsNetwork/lorawan-stack #1723: Problems with management remote VPS server + static IP (ttn-lw-cli + webgui)

ecities, 2019-12-12:
#### Summary

I'm trying to install most recent version of the stack onto **ovh.eu** VPS server (debian 9 stretch) in order to direct connection to our cloud system "on the same dev".

Following your getting started manual (it looks ok when we do **docker-compose up**).

We stuck on trying to initiate configuration (**ttn-lw-cli login** + WEB authorization in the internet browser).

The page always redirect to **localhost**.

...

#### Steps to Reproduce

1. Complete installation of stack on remote server machine IP: **200.200.200.200** in this example via SSH (no desktop and without physical access to local screen). Probably can be reproduced on local server via remote installation without desktop.
2. Running stack successfully (docker-compose up)
3. Replacing most 'localhost', '127.0.0.1' in docker-compose.yml into physical ip address **200.200.200.200**
4. **ttn-lw-client login** alone or with IP (200.200.200.200) parameters
```
ttn-lw-cli login \
      	--network-server-grpc-address 200.200.200.200:8884 \
	--oauth-server-address https://200.200.200.200:8885/oauth \
	--allow-unknown-hosts \
      	--insecure \
      	--join-server-grpc-address 200.200.200.200:8884 \
      	--qr-code-generator-grpc-address 200.200.200.200:8884 \
      	--application-server-grpc-address 200.200.200.200:8884 \
	--device-claiming-server-grpc-address 200.200.200.200:8884 \
      	--device-template-converter-grpc-address 200.200.200.200:8884 \
      	--gateway-server-grpc-address 200.200.200.200:8884 \
      	--identity-server-grpc-address 200.200.200.200:8884 

```

5. Web autorisation always fails and returns to localhost

 ttn-lw-stack config
                                           --as.device-kek-label=""
                                        --as.interop.blob.bucket=""
                                          --as.interop.blob.path=""
                                      --as.interop.config-source=""
                                          --as.interop.directory=""
                                                 --as.interop.id=""
                                                --as.interop.url=""
                                                  --as.link-mode="all"
                                                --as.mqtt.listen=":1883"
                                            --as.mqtt.listen-tls=":8883"
                                        --as.mqtt.public-address="localhost:1883"
                                    --as.mqtt.public-tls-address="localhost:8883"
                                        --as.webhooks.queue-size="16"
                                            --as.webhooks.target="direct"
                               --as.webhooks.templates.directory=""
                           --as.webhooks.templates.logo-base-url=""
                                     --as.webhooks.templates.url=""
                                           --as.webhooks.timeout="5s"
                                           --as.webhooks.workers="16"
                                        --blob.aws.access-key-id=""
                                             --blob.aws.endpoint=""
                                               --blob.aws.region=""
                                    --blob.aws.secret-access-key=""
                                        --blob.aws.session-token=""
                                          --blob.gcp.credentials=""
                                     --blob.gcp.credentials-file=""
                                          --blob.local.directory="./public/blob"
                                                 --blob.provider="local"
                                           --cache.redis.address=""
                                          --cache.redis.database="0"
                                --cache.redis.failover.addresses=""
                                   --cache.redis.failover.enable="false"
                              --cache.redis.failover.master-name=""
[... 247 lines cut ...]
```
~# ttn-lw-stack version
The Things Stack for LoRaWAN: ttn-lw-stack
```
Version:             3.3.2
Build date:          2019-12-04T10:57:50Z
Git commit:          0b1a4d7b3
Go version:          go1.13.4
OS/Arch:             linux/amd64

```

...

#### How do you propose to implement this?

**Is there another way to enable remote authorization without opening remote web page?** eg. **passing ttn-lw-cli login + user/password on the console?** to move on immediately

Fix redirection from localhost to the current server (relative paths).

...

#### Can you do this yourself and submit a Pull Request?

...

htdvisser, 2019-12-12:
Have you seen https://thethingsstack.io/v3.3.2/guides/getting-started/cli/#login ?

Specifically the part that says

> During the login procedure, the CLI starts a webserver on `localhost` in order to receive the OAuth callback after login. If you are running the CLI on a machine that is not `localhost`, you can pass the `--callback=false` flag. This will allow you to perform part of the OAuth flow on a different machine, and copy-paste a code back into the CLI.

ecities, 2019-12-12:
Hi, @htdvisser, Thanks for reply . I saw it .... but it seems I didn't process this information ;) .
I checked all parameters including **--callback false** (without **=** char) as other parameters are supplied.

Your solution,  move it a little forward - into certificates issue.
I have generated certificate with openssl.

ecities, 2019-12-12:
Can the certificate be generated to work remotelly (By public IP) or (localhost) for 
ttn-lw-cli commands for adding gateways, devices, and so on.

htdvisser, 2019-12-12:
If you're setting up The Things Stack on a server with a public IP, you should just point a (sub)domain to it and use our builtin support for requesting and using certificates from Let's Encrypt.

I recommend you to spin up a fresh new server, point some subdomain to it and see if this works for you: https://gist.github.com/htdvisser/4503f30699308a7fd9e0aa3ebc2f3eb4

closed 2019-12-12 by htdvisser