TheThingsNetwork/lorawan-stack #1736: Sending MQTT downlink not working

ktsolakas, 2019-12-17:
#### Summary
I am able to send a downlink to a device through cli using the following command

ttn-lw-cli applications downlink replace --application-id meshed-nas --device-id nas-38dc6 --frm-payload 01FE00 --f-port 60

On the other hand when i try to send a downlink via mqtt client the server logs don't show any downlinks being generated 

#### Steps to Reproduce
1. Create API key for mqtt_client and "Grant all current and future rights"
2. Send downlink via mqtt
mosquitto_pub -h enterprise.meshed.com.au -p 1883 -u meshed-nas@tti-enterprise -P XXXX.W2PVS2QBDKFS5TBRIT6XXXXXXXXXXXXXXXXXXXX.ENFHJVJ2BKZYF3V5WVSDCD6Q5VKT7ZJNXXXXXXXXXXXXXXXXXXXX -t 'v3/meshed-nas/devices/nas-38dc6/down/push' -m '{"downlinks":[{"f_port":60,"frm_payload":"Af5k"}]}'

#### What do you see now?
Logs from the server

Dec 17 05:53:12 ip-10-0-0-251.ap-southeast-2.compute.internal tti-lw-stack[14910]: INFO Connected                                application_uid=meshed-nas@tti-enterprise namespace=applicationserver/io/mqtt remote_addr=13.55.109.245:56677 tenant_id=tti-enterprise
Dec 17 05:53:12 ip-10-0-0-251.ap-southeast-2.compute.internal tti-lw-stack[14910]: DEBUG Subscribed                               application_uid=meshed-nas@tti-enterprise namespace=applicationserver/io/mqtt remote_addr=13.55.109.245:56677 tenant_id=tti-enterprise
Dec 17 05:53:12 ip-10-0-0-251.ap-southeast-2.compute.internal tti-lw-stack[14910]: DEBUG Read publish packet                      client_id=mosq-uoW3w3UBAxyqVRcQGi namespace=applicationserver/io/mqtt username=meshed-nas@tti-enterprise
Dec 17 05:53:12 ip-10-0-0-251.ap-southeast-2.compute.internal tti-lw-stack[14910]: DEBUG Read disconnect packet                   client_id=mosq-uoW3w3UBAxyqVRcQGi namespace=applicationserver/io/mqtt username=meshed-nas@tti-enterprise
Dec 17 05:53:12 ip-10-0-0-251.ap-southeast-2.compute.internal tti-lw-stack[14910]: INFO Disconnected                             application_uid=meshed-nas@tti-enterprise namespace=applicationserver/io/mqtt remote_addr=13.55.109.245:56677 tenant_id=tti-enterprise
Dec 17 05:53:12 ip-10-0-0-251.ap-southeast-2.compute.internal tti-lw-stack[14910]: DEBUG Done sending upstream messages           application_uid=meshed-nas@tti-enterprise error=EOF namespace=applicationserver/io/mqtt remote_addr=13.55.109.245:56677 tenant_id=tti-enterprise
Dec 17 05:53:12 ip-10-0-0-251.ap-southeast-2.compute.internal tti-lw-stack[14910]: DEBUG Unsubscribed                             application_uid=meshed-nas@tti-enterprise namespace=applicationserver/io/mqtt remote_addr=13.55.109.245:56677 tenant_id=tti-enterprise

#### What do you want to see instead?
Downlink to be delivered to the end-device

#### Environment
TTN version 3.3.2
mosquitto_pub version 1.6.8 running on libmosquitto 1.6.8.

#### How do you propose to implement this?
Do not know
#### Can you do this yourself and submit a Pull Request?
Nope

johanstokking, 2019-12-17:
@ktsolakas looks like you're missing the tenant ID in the topic, i.e. it has to be `meshed-nas@tti-enterprise` (depending on the tenant ID). Please see MQTT under Integrations in the Console; the user name is what the part should be in the topic.

ktsolakas, 2019-12-18:
@johanstokking thnx, that did the trick!

johanstokking, 2019-12-18:
There are a few things we can improve in our documentation. Will do that.

closed 2019-12-18 by johanstokking