TheThingsNetwork/lorawan-stack #1737: PubSub MQTT input field validation is too strict

reissjason, 2019-12-17:
#### Summary
When configuring an MQTT PubSub Integration valid entries for client-id or username are blocked by UI validation. Using http://www.dioty.co/mqtt provides an email address for username. This was not allowed. client-id or username containing uppercase letters are also not allowed.

#### Steps to Reproduce

1. Select an Application
2. Add a new PubSub Integration
3. Select MQTT as type
4. Enter "aaaBBB" as client-id or username

#### What do you see now?

[image]

#### What do you want to see instead?

Less restrictive validation should be used.

#### Environment

v3.3.2

#### How do you propose to implement this?

Review validation requirements.

#### Can you do this yourself and submit a Pull Request?

No

htdvisser, 2019-12-18:
This seems to be a restriction imposed by the frontend, as the backend only has a maximum length defined.

htdvisser, 2019-12-18:
Source: `pkg/webui/console/components/pubsub-form/validation-schema.js`

PR #2208 (2020-03-23): Fix too restrictive MQTT client validation
#### Summary
Closes #1737 

#### Changes
- Add new regex to check that strings contain no spaces
- Update pubsub form validation schema

#### Notes for Reviewers
Had trouble finding good info on what exactly are validation rules for mqtt client ids and usernames, but it seems to be quite permissive. So I went with disallowing spaces only.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to t [... cut]
closed 2020-03-23 by kschiffer