TheThingsNetwork/lorawan-stack #1740: Unable to edit existing PubSub integration

reissjason, 2019-12-17:
#### Summary
PubSub Integration screen displays an error. MQTT and NATS integrations both display the error.

...

#### Steps to Reproduce

1. Create a new PubSub integration with dummy values
2. Try to edit the created PubSub integration.
3. ...

#### What do you see now?

[image]

#### What do you want to see instead?

The edit form should be displayed.

#### Environment

v3.3.2

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

No

htdvisser, 2019-12-18:
Error in Devtools: `ReferenceError: "PageTitle is not defined"`

kschiffer, 2019-12-23:
This has been fixed already via #1603.

closed 2019-12-23 by kschiffer