TheThingsNetwork/lorawan-stack #1748: Allow configuring class B on gateways

rvolosatovs, 2019-12-18:
#### Summary

Class B-related options should be available and configurable on gateways 

References #19 https://github.com/TheThingsNetwork/lorawan-stack/pull/1747

#### Why do we need this?

To generate gateway configurations in GCS including class B configuration

#### What is already there? What do you see now?

No class B configuration on gateways

#### What is missing? What do you want to see?

Class B gateway options configurable via both console as well as CLI

#### How do you propose to implement this?

1. Add required fields to proto (@rvolosatovs)
2.
	- [ ] Allow configuring fields from 1. via CLI (@rvolosatovs)
	- [ ] Allow configuring fields from 1. via console (@bafonins) 
	- [ ] Use the fields from 1. in GCS (@rvolosatovs)

#### Can you do this yourself and submit a Pull Request?

yes, partially

kschiffer, 2021-08-16:
@KrishnaIyer can you give an update on this?

bafonins, 2021-08-18:
@johanstokking should we prioritize this?

johanstokking, 2021-08-19:
Not for now.