TheThingsNetwork/lorawan-stack #1749: Random dev_nonce is blocked for MAC_1_0_3

virtualguy, 2019-12-18:
#### Summary

Devices using MAC 1.0.3 should are getting denied join requests when using a randomised dev nonce
...

#### Steps to Reproduce

1. Create OTA device and configure using 
      --lorawan-phy-version 1.0.3-a
      --lorawan-version 1.0.3

2. Join using dev nonce = 5
3. Attempt to re join using dev nonce = 4
4. Monitor stack logs for dev_nonce_too_small

#### What do you see now?

  INFO Cluster-local Join Server did not accept join-request dev_addr=00CB0752 dev_eui=06FDEB37F25B0974 device_uid=__REDACTED__ error=rpc error: code = InvalidArgument desc = error:pkg/joinserver:dev_nonce_too_small (DevNonce is too small) grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs join_eui=__REDACTED__ m_type=JOIN_REQUEST major=LORAWAN_R1 namespace=networkserver received_at=2019-12-18 12:25:58.798582771 +0000 UTC request_id=01DWCFHNREE006JW29MW1WX8GV

#### What do you want to see instead?

Successful join

#### Environment

lorawan_stack v 3.3.2
...

#### How do you propose to implement this?

Only restrict the dev nonce sequence for MAC 1.1

#### Can you do this yourself and submit a Pull Request?

Yes

virtualguy, 2019-12-18:
See https://github.com/TheThingsNetwork/lorawan-stack/pull/1750 for suggested changes

htdvisser, 2019-12-19:
We actually changed this because of the errata for LoRaWAN 1.0.3 that requires incementing DevNonces, see https://github.com/TheThingsNetwork/lorawan-stack/issues/743

htdvisser, 2019-12-19:
@johanstokking already reached out to the Technical Committee. Let's block this issue until we get some clarification from them.

virtualguy, 2019-12-19:
Ah I see, that would explain the code change. My understanding (from the 1.0.3 spec) is that 1.0.2 and 1.0.3 we completely interchangeable for Class A and Class C devices. As a work around we can just configure for 1.0.2.

Surely if it is an errata though the correct thing to do would be applied the changes to a new MAC version of 1.0.3-b

johanstokking, 2019-12-20:
Yes, you are totally correct about the 1.0.3 spec and as such, our code is wrong. So I will be approving your pull request, and I wait for the LoRa Alliance to publish the errata and communicate how to deal with this breaking change.

The errata is there, it got accepted by the Technical Committee in February last year, but it never got published.

The errata is normative, saying that the DevNonce SHALL increment. That is compatible with "random" DevNonces (i.e. incrementing is still different), but not the other way around. So we probably end up with a checkbox for requiring incrementing DevNonces, even though the errata is normative.

rvolosatovs, 2020-01-06:
Closed by https://github.com/TheThingsNetwork/lorawan-stack/pull/1780

closed 2020-01-06 by rvolosatovs