TheThingsNetwork/lorawan-stack #1757: absolute_time field is not observed

virtualguy, 2019-12-19:
#### Summary

When publishing a downlink to a multicast group using the absolute_time field, the message is emitted nearly instantly by the gateway (approx 160ms latency). 

#### Steps to Reproduce

1. Schedule a downlink using the absolute_time field via MQTT a few seconds in advance
2. Watch the gateway, node or SDR for logs that the message was sent earlier than scheduled

#### What do you see now?

Example downlink:
"Sending TTN V3 multicast. topic: v3/halter/devices/e260413e-1685-4b94-9263-31b2f4d7c083/down/replace. Payload: {\"downlinks\":[{\"f_port\":1,\"frm_payload\":\"ChMKEQoPEg0IAxIFDbQiy10YASAH\",\"priority\":\"NORMAL\",\"absolute_time\":\"2019-12-12T01:30:18.000Z\",\"class_b_c\":{\"gateways\":[{\"gateway_ids\":{\"gateway_id\":\"eui-00800000a000530e\"}},{\"gateway_ids\":{\"gateway_id\":\"wirnet\"}}]}}]}"

#### What do you want to see instead?

Message emitted by the gateway at precisely the time specified

#### Environment

...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

rvolosatovs, 2019-12-19:
@virtualguy the absolute time in your example is in the past, which would explain the immediate scheduling, can you confirm that this happens also when the timestamp is in the future?

johanstokking, 2019-12-19:
@virtualguy the problem is that `absolute_time` lives in the `class_b_c` object;

Here's a snippet of a working example using absolute time scheduling over class C:

https://github.com/TheThingsNetwork/lorawan-stack-example-clock/blob/master/controller/src/App.ts#L64-L73

rvolosatovs, 2019-12-19:
@johanstokking why does this not cause an error? I remember @elsalahy had a very similar issue, where he made a typo and the behavior was not what expected.
This is very confusing to the user.
I guess the error handling is difficult with MQTT, so the AS just silently ignores malformed fields? In any case, IMO correct behavior would be to at least silently drop such malformed downlinks, but those really should cause some kind of observable error - in case it's difficult to do with MQTT - maybe an event?

johanstokking, 2019-12-20:
The input is valid JSON, so we can't fail. We don't validate JSON schema in incoming MQTT and webhook downlink messages. We can do that, but we have to bump major, see [compatibility commitment rule #1](https://github.com/thethingsnetwork/lorawan-stack/#commitments-and-releases).

So this gets interpreted as any JSON unmarshaling; it's an unknown field that gets discarded. In this case, `class_b_c` is optional, so the downlink message is not invalid.

I don't see a reason to fail, and I don't see room for failing in V3.

johanstokking, 2019-12-20:
@virtualguy can you confirm that it works this way?

rvolosatovs, 2019-12-20:
@johanstokking see https://github.com/TheThingsNetwork/lorawan-stack/issues/1771

closed 2020-02-07 by johanstokking