TheThingsNetwork/lorawan-stack #1783: Set up Webhook Templates repository

adriansmares, 2019-12-23:
#### Summary

We should create a Webhook Templates repository under TheThingsNetwork's GitHub organization.

#### Why do we need this?

The Webhook templates repository will contain descriptions and settings for webhook-based integrations which require both an HTTP endpoint and certain settings such as authorization tokens or which region to use, descriptions and names.

#### What is already there? What do you see now?

We have currently the following:
 - Webhook templates definitions and retrieval services in `applicationserver_web.proto`
 - Webhook templates storage support in the AS under `pkg/applicationserver/io/web` (listing / retrieving templates from a fetcher using `pkg/fetcher`)
 - The AS already exposes the template related RPCs when a fetcher is configured (see [here](https://github.com/TheThingsNetwork/lorawan-stack/blob/3e2a02b31bab5bd253ad8c7766bf6b0540cab98b/pkg/applicationserver/io/web/templates.go#L56-L77) for the specific configuration - nothing different from other fetchers).

#### What is missing? What do you want to see?

There is no default template repository that we can configure. 

We'll also need #945 for the actual tiles of the templates in the Console (i.e. instantiation of the template into am `ApplicationWebhook` instance) and #1782 for the documentation of the format.

#### How do you propose to implement this?

Create the repository `TheThingsNetwork/lorawan-stack-integrations` and use it as a default in the The Things Stack.

#### Can you do this yourself and submit a Pull Request?

Yes. cc @johanstokking

adriansmares, 2020-01-08:
Blocked by #1782, #1699.

adriansmares, 2020-07-09:
@nejraselimovic has been working on setting up the repository at https://github.com/TheThingsIndustries/lorawan-webhook-templates . I find the contents to be fine, only that we should probably rebase `master` to squash the fixups.

@johanstokking could you please review the current state of `master` in that repository, and if everything is fine maybe we should move it under `TheThingsNetwork` ?

johanstokking, 2020-07-10:
@nejraselimovic please file pull request from `develop` and request review from @adriansmares and myself

closed 2020-07-17 by adriansmares