TheThingsNetwork/lorawan-stack #1789: Can't join Things Kickstarter Gateway to server

loganmc10, 2019-12-24:
#### Summary

I followed this guide: https://thethingsstack.io/v3.4.0/guides/connecting-gateways/thethingskickstartergateway/ to join my gateway to my TTN stack server. The gateway appears to be attempting to join, this is what shows up on the console of the server:
```
duration=18.325µs error=error:pkg/errors/web:unknown (Not Found) message=Not Found method=GET namespace=web remote_addr=x.x.x.x:5256 request_id=01DWWA895FS8RVNWRQC1PWRPQP status=404 url=/api/v2/gateways/ttn-ks-gateway?filter=ttn
```

#### Steps to Reproduce

Attempt to join a kickstarter gateway to a TTN stack server

#### Environment

TTN stack 3.4.0, running on Ubuntu VM in AWS

KrishnaIyer, 2019-12-24:
Thanks @loganmc10. Are you running the stack using `start`. The GCS is not started by default so you need to run the stack with `start all`.

loganmc10, 2019-12-24:
I run the stack using docker-compose, so I run ```docker-compose up```

loganmc10, 2019-12-24:
I see in my docker file:
```
  stack:
    image: thethingsnetwork/lorawan-stack:3.4.0
    entrypoint: ttn-lw-stack
    command: start
```

You're saying I should change the command to ```start all```?

KrishnaIyer, 2019-12-24:
Then please update your docker compose file to use `command: start all`

loganmc10, 2019-12-24:
Ok perfect I have it connected now, a couple questions around documentation:

where is the ```start all``` requirement documented? Every document I saw just lists "start"

Secondly, it appears like port 1881/8881 is used for MQTT? For instance I see:
```
- TTN_LW_GS_MQTT_V2_PUBLIC_ADDRESS=thethings.example.com:1881
- TTN_LW_GS_MQTT_V2_PUBLIC_TLS_ADDRESS=thethings.example.com:8881
- TTN_LW_GCS_THE_THINGS_GATEWAY_DEFAULT_MQTT_SERVER=mqtts://thethings.example.com:8881
```
In the example docker file, however 1881/8881 is not listed in the "ports" section. I had to add it to the ports section before the gateway would join. Maybe this was just a coincidence? I might have to test it again without that in the ports section to be sure

KrishnaIyer, 2019-12-24:
Yeah you're right. This is not documented.

We'd be more than happy if you create a PR for this. I'd recommend creating a sub-section under `Running The Things Stack` in `guides/getting-started/running-the-stack/`.

> In the example docker file, however 1881/8881 is not listed in the "ports" section. I had to add it to the ports section before the gateway would join. Maybe this was just a coincidence? I might have to test it again without that in the ports section to be sure

This is not a functional change but serves as a reference documentation. Feel free to add this in the doc as well.  You can add `8886` to this list since we've not documented that as well.

htdvisser, 2020-01-23:
PR #1894 and #1895 should close this issue.

closed 2020-01-25 by loganmc10