TheThingsNetwork/lorawan-stack #1807: Fail to update the gateway eui via cli

virtualguy, 2020-01-06:
#### Summary
When trying to just set the gateway eui the field wont update. A work around it to set an additional field (such as --name) at the same time and then the command takes affect.

#### Steps to Reproduce
1. ttn-lw-cli gateway create example_gateway
2. ttn-lw-cli gateway update example_gateway --gateway-eui 00800000a0999999
3. Observe that the gateway eui is not set

#### What do you see now?
ttn-lw-cli gateway update example_gateway --gateway-eui 00800000a0999999
  WARN No fields selected, won't update anything

#### What do you want to see instead?
ttn-lw-cli gateway update example_gateway --gateway-eui 00800000a0999999
{
  "ids": {
    "gateway_id": "example_gateway",
    "eui": "00800000a0999999"
  },
  "created_at": "2020-01-06T23:07:14.624Z",
  "updated_at": "2020-01-06T23:11:44.343Z",
  "version_ids": {

  }
}

#### Environment
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.4.1
Build date:          2019-12-30T12:14:16Z
Git commit:          283798f88
Go version:          go1.13.5
OS/Arch:             linux/amd64

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

neoaggelos, 2020-01-07:
Thanks @virtualguy for reporting this. This is indeed a bug with the cli. I have tracked the cause and currently working on a solution.

virtualguy, 2020-01-07:
Thanks @neoaggelos to confuse matters if I specify an eui then it reponds back with the eui set but does not actually update the database. So the above mentioned workaround does not help

PR #1808 (2020-01-08): Fix `gateways update` command refusing to update EUI

#### Summary

Closes #1807 

#### Changes

- Add the `ids.eui` path when GatewayEUI has been set

#### Notes for Reviewers

- Perhaps there are other cli update commands with similar issues?

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [ ] Testing: The changes are covered with unit tests. The changes are tested manually as well.
- [... cut]
closed 2020-01-08 by neoaggelos