TheThingsNetwork/lorawan-stack #1814: Pagination does not work for search requests

kschiffer, 2020-01-07:
#### Summary
The `page` prop is not working properly for `SearchEndDevicesRequest`.

Blocking #1803 

#### Steps to Reproduce
1. Create an app with the name `test-app`
2. Create two devices with the IDs `device-one` and `device-two`
3. Run `GET http://localhost:1885/api/v3/search/applications/test-app/devices?page=2&limit=1&id_contains=device`
4. Observe an empty response

Same holds true for all other search endpoints.

#### What do you see now?
Page option > 1 yielding emtpy responses

#### What do you want to see instead?
Pagination working properly for search endpoints

#### Environment
v3.4.1

#### How do you propose to implement this?
Haven't checked the backend code, so cannot make any suggestions.

#### Can you do this yourself and submit a Pull Request?
Please take a look @htdvisser or @bafonins

htdvisser, 2020-01-09:
@kschiffer it would save a lot of time if you could add backend logs (with log level debug) so that we can see the queries that are executed.

kschiffer, 2020-01-09:
Sorry.

Request: 
```
curl -X GET \
  'http://localhost:1885/api/v3/search/applications?page=1&limit=1&id_contains=app&field_mask=name,description' \
  -H 'Accept: application/json, text/plain, */*' \
  -H 'Accept-Encoding: gzip, deflate, br' \
  -H 'Authorization: Bearer MFRWG.3K3AKUQA7TFKRRVPGRMSVHYFZLZR35OEET7TOEY.PTZBTQCRLK7HJFW4W6BMKVZLAXSPRT2I3VNLCCQMOHUKDOCSVFCA,Bearer MFRWG.SBBCMHVFHLHSPCEYYDWXE37XCAPTVDTODV632YA.3F7D6BM3THYEFUGF4ADM3SKAHIC5HZEIWECX2BDI43TUZZTQJKWA' \
  -H 'Host: localhost:1885' \
```

Relevant backend logs: 
```
 DEBUG Run database query                       duration=8ms grpc_method=SearchApplications grpc_service=ttn.lorawan.v3.EntityRegistrySearch namespace=db query=SELECT "access_tokens".*, "accounts"."uid" AS "friendly_user_id", "clients"."client_id" AS "friendly_client_id" FROM "access_tokens" JOIN "accounts" ON "accounts"."account_type" = 'user' AND "accounts"."account_id" = "access_tokens"."user_id" JOIN "clients" ON "clients"."id" = "access_tokens"."client_id" WHERE ("access_tokens"."token_id" = $1) request_id=01DY565YA6G4G2TPXCPNA7XJDQ rows=1 source=oauth_store.go:271 values=[SBBCMHVFHLHSPCEYYDWXE37XCAPTVDTODV632YA]
 DEBUG Run database query                       duration=8ms grpc_method=SearchApplications grpc_service=ttn.lorawan.v3.EntityRegistrySearch namespace=db query=SELECT users.id, users.created_at, users.updated_at, admin, state, primary_email_address_validated_at FROM "users" LEFT JOIN accounts ON accounts.account_type = $1 AND accounts.account_id = users.id WHERE "users"."deleted_at" IS NULL AND ((accounts.uid = $2)) ORDER BY "users"."id" ASC LIMIT 1 request_id=01DY565YA6G4G2TPXCPNA7XJDQ rows=1 source=user_store.go:121 values=[user admin]
 DEBUG Run database query                       duration=9ms grpc_method=SearchApplications grpc_service=ttn.lorawan.v3.EntityRegistrySearch namespace=db query=SELECT * FROM "accounts"  WHERE "accounts"."deleted_at" IS NULL AND (("account_id" IN ($1) AND "account_type" = $2)) ORDER BY "accounts"."id" ASC request_id=01DY565YA6G4G2TPXCPNA7XJDQ rows=1 source=user_store.go:121 values=[cc38a4c0-1b37-46de-8aa2-2b789eb505a2 user]
 DEBUG Run database query                       duration=13ms grpc_method=SearchApplications grpc_service=ttn.lorawan.v3.EntityRegistrySearch namespace=db query=SELECT id, created_at, updated_at, client_id, state FROM "clients"  WHERE "clients"."deleted_at" IS NULL AND ((client_id = $1)) ORDER BY "clients"."id" ASC LIMIT 1 request_id=01DY565YA6G4G2TPXCPNA7XJDQ rows=1 source=client_store.go:113 values=[console]
 DEBUG Run database query                       duration=8ms grpc_method=SearchApplications grpc_service=ttn.lorawan.v3.EntityRegistrySearch namespace=db query=SELECT "applications"."application_id" AS "friendly_id" FROM "applications"  WHERE "applications"."deleted_at" IS NULL AND (("applications"."application_id" LIKE $1)) ORDER BY friendly_id ASC LIMIT 1 OFFSET 0 request_id=01DY565YA6G4G2TPXCPNA7XJDQ rows=1 source=entity_search.go:122 values=[%app%]
 DEBUG Run database query                       duration=16ms grpc_method=SearchApplications grpc_service=ttn.lorawan.v3.EntityRegistrySearch namespace=db query=SELECT count(*) FROM "applications"  WHERE "applications"."deleted_at" IS NULL AND (("applications"."application_id" LIKE $1)) request_id=01DY565YA6G4G2TPXCPNA7XJDQ rows=0 source=pagination.go:60 values=[%app%]
 DEBUG Run database query                       duration=12ms grpc_method=SearchApplications grpc_service=ttn.lorawan.v3.EntityRegistrySearch namespace=db query=SELECT "memberships"."rights" FROM "memberships"  WHERE ("memberships"."account_id" = (SELECT "accounts"."id" FROM "accounts"  WHERE "accounts"."deleted_at" IS NULL AND (("accounts"."account_type" = 'user' AND "accounts"."uid" = $1)))) AND ("memberships"."entity_type" = 'application' AND "memberships"."entity_id" = (SELECT "applications"."id" FROM "applications"  WHERE "applications"."deleted_at" IS NULL AND ((application_id = $2)))) ORDER BY "memberships"."id" ASC LIMIT 1 request_id=01DY565YA6G4G2TPXCPNA7XJDQ rows=1 source=membership_store.go:192 values=[admin test-app]
 DEBUG Run database query                       duration=10ms grpc_method=SearchApplications grpc_service=ttn.lorawan.v3.EntityRegistrySearch namespace=db query=SELECT id, created_at, updated_at, application_id, description, name FROM "applications"  WHERE "applications"."deleted_at" IS NULL AND ((application_id = $1)) ORDER BY application_id ASC LIMIT 1 OFFSET 0 request_id=01DY565YA6G4G2TPXCPNA7XJDQ rows=1 source=application_store.go:94 values=[test-app]
  INFO Finished unary call                      duration=181.833517ms grpc_method=SearchApplications grpc_service=ttn.lorawan.v3.EntityRegistrySearch namespace=grpc request_id=01DY565YA6G4G2TPXCPNA7XJDQ
```

htdvisser, 2020-01-09:
This looks really strange. Are you sure the logs correspond to this exact request?

- The request has `page=2`, but the logs show a query with `OFFSET 0`
- The DB logs show `rows=1`, but your response is an empty `{}`

kschiffer, 2020-01-09:
Yeah, sorry. The log was not the one corresponding to the request. I've updated my comment.

kschiffer, 2020-01-20:
Unfortunately, this is still not resolved. The request above still yields the same result.
Appears like I did not test your PR thoroughly enough. Mea culpa 🙏

PR #1856 (2020-01-15): Reset pagination in second stage of search

#### Summary

This is a quickfix that closes #1814 by resetting pagination for the second stage of search. The correct page is already selected by the first stage, so we can just return all results.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [x] Testing: The changes are covered with unit tests. The changes are tested manually as well.
- [ [... cut]
PR #1901 (2020-01-23): Remove obsolete pagination fields from headers

#### Summary

This pull request removes the obsolete limit and offset from the rpc metadata. These fields are always part of the request messages, and extracting them from the header as well was causing issues with pagination.

This should fix #1814.

#### Changes

- Remove `Limit` and `Offset` fields from `rpcmetadta.MD`
- Remove all remaining usage

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuratio [... cut]
closed 2020-01-23 by htdvisser