TheThingsNetwork/lorawan-stack #1824: Support Multiple Frequency Plans in Scheduler

KrishnaIyer, 2020-01-08:
#### Summary

Support Multiple Frequency Plans in Scheduler (refs #761)

#### Why do we need this?

Enable downlinks on multiple concentrators.

#### What is already there? What do you see now?

Branch `feature/761-multi-fplan-gtw` contains the rest of the items discussed in #761.

#### What is missing? What do you want to see?

Use the supported `FrequencyPlans` struct in the scheduler and use the `frequency_plan_id` as part of the `TxRequest` to choose the correct downlink parameters.

#### Environment

ttn-lw-stack v3.4.1

#### How do you propose to implement this?

Better left to the expert.

#### Can you do this yourself and submit a Pull Request?

Can review and test if necessary.

johanstokking, 2020-01-09:
Can you see if you can pick this up? Would be great as a separate PR and not part of #1826.

KrishnaIyer, 2020-01-09:
Ok. I'll take this.

KrishnaIyer, 2020-01-14:
To summarise our discussions here:
- For now the FPs have to be from the same band.

As for combining frequency plans in the scheduler
- The timeOffAir parameter should be the same (same value or unset) across the FPs?
- As for sub bands, it would be a **union** of the frequencies right? The sub-bands encapsulate the difference in duty cycling so creating sub-bands based on the union of the FPs should work right?

johanstokking, 2020-01-14:
> * The timeOffAir parameter should be the same (same value or unset) across the FPs?

Yes. If they don't match we can safely error.

> * As for sub bands, it would be a **union** of the frequencies right? The sub-bands encapsulate the difference in duty cycling so creating sub-bands based on the union of the FPs should work right?

That would be best. There are three scenarios then;

1. There are no FR sub-bands defined, which is great
2. The same sub-band for all FPs is configured, this is also good
3. The sub-bands of the given frequency plans do not overlap, this is also fine
4. The sub-bands of the given frequency plans are not the same and overlap. This is not good

KrishnaIyer, 2020-01-16:
closed via #1859

PR #1859 (2020-01-16): Scheduler with Multiple frequency plans

#### Summary

Closes #1824  Add multiple frequency plans support for the Scheduler

#### Changes

- Ensure that fps have the same `TimeOffAir` value.
- Create a Union of sub bands and check for overlap.
- Check if new emissions respect the DwellTimeRestriction of at least one frequency plan.
- Tests

#### Notes for Reviewers

This PR is targeted to the `feature/761-multi-fplan-gtw` branch.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existi [... cut]
closed 2020-01-16 by KrishnaIyer