TheThingsNetwork/lorawan-stack #1825: Advanced entity search (in console)

kschiffer, 2020-01-08:
#### Summary
Entity search in the console is currently limited to IDs only (#1803). We should enhance search capabilities for advanced use cases.

#### Why do we need this?
Users with a lot of entities might want to find specific entities based on properties other than the ID.

#### What is already there? What do you see now?
- Entity search by ID in the console
- Search RPCs of the entity registry
- Inclusive combination of search criteria (by "logic and" operator)

#### What is missing? What do you want to see?
- Advanced search allowing for a combination of search criteria and search criteria other than the ID
  - ability to combine search criteria with "logic or" in the RPCs (@htdvisser noted that this might be unfeasible anytime soon – should I open an issue about this still?)
  - Advanced search UI in the console

#### Environment
v3.4.1

#### How do you propose to implement this?
The solution might be either a dedicated search page or an advanced search panel as part of the table component.

#### Can you do this yourself and submit a Pull Request?
Yes.

htdvisser, 2021-10-12:
Cross-posting my comment from #4056:

We've been discussing multiple things:

1. The "advanced search" functionality where users can search on fields other than the ID, which would be implemented in the Console as something like GMail's advanced search:
 
[image]

This would just fill the fields of the existing search requests.

This is basically what's tracked in #1825.

---

2. The "universal search" functionality where a single Spotlight-like search input (in the top of the Console) would magically search for all applications, gateways, end devices, etc. that contain the given text in any of their fields (id, euis, name, description, attributes, ...).

I don't think we have an issue for this yet.

pgalic96, 2021-11-03:
> @kschiffer @pgalic96 can we maybe merge all of these issues into this one?
> 
> Searching on the Name field: #4056 Searching on the Gateway EUI field: #3327 Searching on the Attributes field: #2586 Also related: #1910

I am fine with this, but does it mean we should halt work on the separate search issues? We currently need UX/UI design for "advanced search" and/or separate search fields for each field that would be searchable. cc @kschiffer 
I have https://github.com/TheThingsNetwork/lorawan-stack/issues/3327 assigned in this milestone but I think we first need a decision on how we want to approach this. I suggest we start immediately on the "advanced" gmail-like approach.

KrishnaIyer, 2024-10-22:
The search currently does use the `query` field which searches for EUI, ID, Name and Description. This is sufficient for most use cases so the issue can be closed.

closed 2024-10-22 by KrishnaIyer