TheThingsNetwork/lorawan-stack #1831: Messages stop being sent to connected mqtt clients when re-linking application

virtualguy, 2020-01-09:
#### Summary
When a existing mqtt or grpc client is connected and the application link is updated (to the same value) the existing client stops receiving messages for all topics. The client does continue to stay connected and gets keep alive messages (pingresp). On restarting the mqtt client the traffic returns to normal

#### Steps to Reproduce
1. Create an application as per the getting started guide
2. Create api key and use to connect via mqtt
3. Validate that uplink messages arrive as expected
4. Re link the application using the same api key as when it was created
5. Observe that uplinks arrive on the web console but not to the mqtt client

The same behaviour happens when using 
ttn-lw-cli applications subscribe

#### What do you see now?
Uplink traffic stops when the application link is re set

#### What do you want to see instead?
Traffic to momentarily stop while the application link is 'unlinked' and the restore for all connected and subscribed clients when the link is restored

#### Environment
Stack and cli version: 3.4.2

#### How do you propose to implement this?

#### Can you do this yourself and submit a Pull Request?

virtualguy, 2020-01-12:
@htdvisser @johanstokking  thoughts?

johanstokking, 2020-01-13:
@adriansmares existing subscriptions on links (like MQTT) need to be restored somehow when the link disappears. I think this has to be initiated by the subscriber, i.e. MQTT frontend, to start watching the link become available while it has clients connected.

PR #1724 (2019-12-12): Clean up documentation
#### Summary
Fixes #1831
Apply changes from https://github.com/TheThingsIndustries/lorawan-stack/pull/1845 belonging to 
 TheThingsNetwork/lorawan-stack.

#### Changes
Replace getting started link on index
Replace component image alt.
Move CLI installation from getting-started/installation to  getting-started/cli
PR #1875 (2020-01-23): Automatically reconnect frontends on link failure

#### Summary

Closes #1831 

#### Changes

- Added a new `io.Server` that attempts to reconnect to the upstream server (Application Server) if the subscription fails.
- Subscriptions now automatically disconnect on link failure (before this PR they would just get stuck).

#### Notes for Reviewers

None.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in  [... cut]
closed 2020-01-23 by adriansmares