TheThingsNetwork/lorawan-stack #1837: Support Generation of router_config for 16/64 Channel UDP gateways

MathieuMonneret, 2020-01-10:
#### Summary

Add support to return a UDP packet forwarder global_conf.json with multiple concentrator support.
see also #761 

#### Why do we need this?

This is to support the configuration of 16 (or more) ch gateways with UDP Ex: Cisco Wireless gateway and Kerlink.
#### What is already there? What do you see now?

`pfconfig` supports only `SX1301_conf` object.

#### What is missing? What do you want to see?

support also `SX1301_array_conf` object.

```
{                                                                         
    "SX1301_array_conf":[                                                 
    {                                                                     
        "board_type": "MASTER",                                           
        "board_rx_freq": 866500000, /* radio RX central frequency in Hz */
        "board_rx_bw": 7000000, /* radio RX bandwidth in Hz */
        "full_duplex": false, /* radio mode */
        "rf_chain_conf":[               
        {                                                                       
            /* RX */                                                            
            "rx_enable": true,                                                  
            "rssi_offset": -202.5,                                              
            "rssi_offset_coeff_a": 20,                                          
            "rssi_offset_coeff_b": 2925,                                        
            /* TX */                                                            
            "tx_enable": true,                                                  
            "tx_freq_min": 863000000,                                           
            "tx_freq_max": 870000000,                                           
            "tx_lut":[                                                          
                { "rf_power":  2, "fpga_dig_gain": 4, "ad9361_atten": 125, "ad93
                { "rf_power":  5, "fpga_dig_gain": 4, "ad9361_atten": 112, "ad93
                { "rf_power":  7, "fpga_dig_gain": 4, "ad9361_atten": 105, "ad93
                { "rf_power":  9, "fpga_dig_gain": 4, "ad9361_atten":  98, "ad93
                { "rf_power": 11, "fpga_dig_gain": 4, "ad9361_atten":  90, "ad93
                { "rf_power": 12, "fpga_dig_gain": 4, "ad9361_atten":  86, "ad93
                { "rf_power": 13, "fpga_dig_gain": 4, "ad9361_atten":  83, "ad93
                { "rf_power": 14, "fpga_dig_gain": 4, "ad9361_atten":  77, "ad93
                { "rf_power": 16, "fpga_dig_gain": 4, "ad9361_atten":  68, "ad93
                { "rf_power": 18, "fpga_dig_gain": 4, "ad9361_atten":  59, "ad93
                { "rf_power": 20, "fpga_dig_gain": 4, "ad9361_atten":  51, "ad93
                { "rf_power": 22, "fpga_dig_gain": 4, "ad9361_atten":  44, "ad93
                { "rf_power": 24, "fpga_dig_gain": 4, "ad9361_atten":  34, "ad93
                { "rf_power": 25, "fpga_dig_gain": 4, "ad9361_atten":  30, "ad93
                { "rf_power": 26, "fpga_dig_gain": 4, "ad9361_atten":  24, "ad93
                { "rf_power": 27, "fpga_dig_gain": 4, "ad9361_atten":  20, "ad93
        }],                                                                     
        "SX1301_conf":[                                                         
        {                                                                       
            "chip_enable": true,                                                
            "chip_center_freq": 865600000,                                      
            "chip_rf_chain": 0,                                                 
            "chan_multiSF_0": { "chan_rx_freq": 864900000, "spread_factor": "7-1
            "chan_multiSF_1": { "chan_rx_freq": 865100000, "spread_factor": "7-1
            "chan_multiSF_2": { "chan_rx_freq": 865300000, "spread_factor": "7-1
            "chan_multiSF_3": { "chan_rx_freq": 865500000, "spread_factor": "7-1
            "chan_multiSF_4": { "chan_rx_freq": 865700000, "spread_factor": "7-1
            "chan_multiSF_5": { "chan_rx_freq": 865900000, "spread_factor": "7-1
            "chan_multiSF_6": { "chan_rx_freq": 866100000, "spread_factor": "7-1
            "chan_multiSF_7": { "chan_rx_freq": 866300000, "spread_factor": "7-1
            "chan_LoRa_std" : { "chan_rx_freq": 866100000, "bandwidth": 250000, 
            "chan_FSK"      : { "chan_rx_freq": 866600000, "bandwidth": 125000, 
        },{                                                                     
            "chip_enable": true,                                                
[... 55 lines cut ...]
        /* Beaconing parameters */                                              
        "beacon_conf":{                                                         
            "period": 128,                                                      
            "freq_hz": 869525000,                                               
            "datarate": 9,                                                      
            "bw_hz": 125000,                                                    
            "power": 14,
            "infodesc": 0                                                       
        }                                                                       
    },                                                                          
    "gps_conf": {                                                               
        "gw_latitude": 0,                                                       
        "gw_longitude": 0,                                                      
        "gw_altitude": 0,                                                       
        "fixed_altitude": false                                                 
    }                                                                           
}                                           
```
#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

@KrishnaIyer @rvolosatovs

htdvisser, 2020-08-24:
Moving to Backlog until we have more demand (👍's)

KrishnaIyer, 2025-04-30:
This was only for a few niche gateways and we don't recommend UDP in general in favor of LoRa Basics Station gateways, which support multi-concentrator frequency plans.

closed 2025-04-30 by KrishnaIyer