TheThingsNetwork/lorawan-stack #1840: Ability to configure gateway schedule delay

neoaggelos, 2020-01-12:
#### Summary

With #1834 gateways have a new `schedule_anytime_delay` field. It is currently configurable using the API and the CLI:

```
$ ttn-lw-cli gateways update gateway1 --schedule-anytime-delay 500ms
```

We should be able to have 

#### Why do we need this?

...

#### What is already there? What do you see now?

```
$ ttn-lw-cli gateways update gateway1 --schedule-anytime-delay 500ms
```

#### What is missing? What do you want to see?

An extra field in the Console gateway settings page, able to configure the `schedule_anytime_delay` field of the gateway.

#### Environment

all

#### How do you propose to implement this?

#### Can you do this yourself and submit a Pull Request?

can test

bafonins, 2020-01-13:
Blocked by https://github.com/TheThingsNetwork/lorawan-stack/pull/1834

> It is not as straightforward as adding a new string field, because schedule_anytime_delay is a duration.

Can we implement this similar to the `JoinEUIPrefixesInput`? Something like this:
[image]
where the second input is of type `number`

cc @kschiffer

kschiffer, 2020-01-13:
Good idea, though I would swap the order: number first, then unit.

johanstokking, 2020-01-13:
Yes, both good ideas

PR #2113 (2020-03-24): Schedule anytime delay in the console

#### Summary

Closes #1840 

#### Changes

- Add new `ScheduleDelayInput` component that is composed of two fields
- Add `schedule_anytime_delay` field when making api request

#### Notes for Reviewers

- Since `schedule_anytime_delay` is returned as a string, I added some logic for checking substrings for unit and value. This was a bit tricky when it came to validating and especially displaying warnings when the value is lower than `130ms` . Initially I wanted to make the `ScheduleDelayInput` more generic, to be able to add it to other places that possib [... cut]
closed 2020-03-24 by pgalic96