TheThingsNetwork/lorawan-stack #1844: Logging out does not work

rvolosatovs, 2020-01-13:
#### Summary

Logging out does not work

#### Steps to Reproduce

1. Log in in the console
2. Log out in the console

#### What do you see now?

After clicking "log out" button I'm redirected to a page with single button "Login", after clicking "Login", I'm logged in again without asking for my username and/or password.

#### What do you want to see instead?

Show the login screen after "log out" button is pressed, i.e. let the user relogin and/or login using different account.

#### Environment

Linux 5.4.8, Firefox 71.0

#### How do you propose to implement this?

I suppose the issue is the cookie? If so, either delete it or mark it as disabled, not sure how this works.

#### Can you do this yourself and submit a Pull Request?

no

kschiffer, 2020-01-13:
That is because you're only logging out from the console as OAuth client, not from the OAuth provider itself. Hence you can log back in without being asked for credentials as your still authenticated for the OAuth provider. To logout from the OAuth provider, you need to go to `/oauth`.
Technically, this is the correct behavior, but we're aware of the UX issue this creates. We will fix this issue as part of #1422

jpmeijers, 2020-02-15:
I agree with @rvolosatovs that this behaviour seems incorrect. I want to click logout so that I can log in as a different user. This is currently not possible using the user expected flow.

jpmeijers, 2020-02-15:
1. Go to https://lns.ttnmapper.org/, log in, and get redirected to https://lns.ttnmapper.org/console
2. Go to https://lns.ttnmapper.org/oauth and click logout
3. You are redirected to https://lns.ttnmapper.org/oauth/login to log in again, but don't log in.
4. Go to https://lns.ttnmapper.org/console without logging in
5. You can access the console without having to log back in.

The login/logout state seems to be different between the console and oath. This is very counter-intuitive.

A workaround seems to be to both log out of the console and also log out of oath to be properly logged out.

kschiffer, 2020-03-24:
Duplicate of #1702; will be resolved via #2148

closed 2020-03-24 by kschiffer