TheThingsNetwork/lorawan-stack #1890: Add "rights" field to user entity

htdvisser, 2020-01-22:
#### Summary

I think we should add an optional "rights" field to the User entity. This field may be edited by admins only, and would restrict the rights that a user has in the network if set.

A configuration option under `is.user-registration` could be added to restrict the rights of new users.

#### Why do we need this?

A network operator may want to prevent users from creating entities in their own account, and only allow creating entities in organizations they have access to. This feature could also be used to prevent users from requesting OAuth clients.

#### What is already there? What do you see now?

- `is.authInfo` gets the user entity when using a User API key or OAuth access token
- `is.getRights` returns the rights of `is.authInfo` when a user lists their rights on their own user.

#### What is missing? What do you want to see?

Intersecting the rights with the `rights` field in the User

#### How do you propose to implement this?

- Add field to `User` API
- Add field to `User` model
- Intersect `userRights` with `rights` of `user` in `is.authInfo`

laurensslats, 2020-06-08:
To clarify, requested feature is to set rights for:
Creating/modifying applications
Creating/deleting devices
Creating/deleting gateways

benolayinka, 2020-07-13:
> To clarify, requested feature is to set rights for:
> Creating/modifying applications
> Creating/deleting devices
> Creating/deleting gateways

And in addition to this, using organizations to manage these permissions for groups of users.
So for example, a "read-only" organization could be used to easily create and group accounts without rights to create anything.

htdvisser, 2020-07-14:
@benolayinka I don't think that is feasible, as it would require changes to our entire rights model.

The easiest way to implement the requirements listed by @laurensslats is to add network-wide configuration to prevent regular (non-admin) users from creating applications/gateways/organizations. Instead, admins will create those, and add users as members on those entities.

@laurensslats ACK?

laurensslats, 2020-07-14:
@htdvisser yes, this sounds perfect.

Let's focus on devices, applications, gateways and organizations. 
Admins can pose restrictions on users to create/modify/delete any of these.

By default, all users should have all the rights when creating an account.

htdvisser, 2020-07-15:
- This would apply to Applications, Gateways, Organizations and OAuth clients. End Device rights are managed through application membership rights, so no need to build something custom for that.
- It would be configuration for the entire network, that applies to all non-admin users.
- The default value for the option would indeed be the same as the current behavior (allowed).

htdvisser, 2020-07-20:
Closed by #2918

PR #2918 (2020-07-16): Add options to restrict create rights for non-admin users

#### Summary

Add options for restricting rights of non-admin users to create entities.

Closes #1890

#### Testing

Unit tests

##### Regressions

N/A

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md`. The target branch is set to `master` if the changes are fully compatible with existing API, database, configuration and CLI. [... cut]
closed 2020-07-20 by htdvisser