TheThingsNetwork/lorawan-stack #1891: Console event log is broken ofr devices and gateways

Oliv4945, 2020-01-22:
#### Summary
Event log in the console is no longer working. It displays 
```
Cannot show events
An unknown error occurred. Please try again later.
```

#### Steps to Reproduce

1. Use HEAD code from master or any Docker build since [this one](https://hub.docker.com/layers/thethingsnetwork/lorawan-stack-dev/50f59ad3e1b68a5d2f98264e28a271b87c8e2cf2/images/sha256-f0d7ad91799941518d122c73c37a19a9c2c43f17a0fb2238d1ea643f0fab9e61)

Older images are working. I do not know the correlation between this image and the Git commits.  

There is no issue with the CLI event stream

htdvisser, 2020-01-23:
Referenced Docker image is from 50f59ad3e1b68a5d2f98264e28a271b87c8e2cf2. 
PR #1870 by @bafonins.

htdvisser, 2020-01-23:
If event streams are indeed broken, this needs to be fixed before the next release

bafonins, 2020-01-23:
@Oliv4945 please reopen if the issue is not resolved in https://github.com/TheThingsNetwork/lorawan-stack/pull/1896

closed 2020-01-23 by bafonins

Oliv4945, 2020-01-23:
@bafonins I confirm that the issue is solved. Thank you