TheThingsNetwork/lorawan-stack #1910: Extend store logic for search results

kschiffer, 2020-01-24:
#### Summary
We need to change the way we handle search results in our redux store. Currently, we treat search request just like regular list requests.

#### Why do we need this?
- The `totalCount` prop we're using will be affected by search requests, resulting in the UI showing wrong values for the total number of entities
- Search requests will trigger a `fetching` toggle for the `GET_*_LIST` actions, resulting in some elements rendering a loading spinner when they don't actually have to

#### What is already there? What do you see now?
- Current state of the store which includes the new pagination logic

#### What is missing? What do you want to see?
A distinction of `LIST_*` and `SEARCH_*` requests

#### Environment
Console @ v3.5.0

#### How do you propose to implement this?
Cannot think this through in detail right now, but maybe adding a `filteredIds` to the pagination store, which will hold the currently filtered entities.

#### Can you do this yourself and submit a Pull Request?
Yes but @bafonins feel free to take over

pgalic96, 2020-01-26:
- regarding the issue with updating entity count: let's just have a new variable that stores the total value of devices (would be computed when the view is mounted). that way it will not depend on the `totalCount` that is used for pagination.

- regarding the fetching issue, my suggestion is to create a new store `SEARCH_*` request that will not trigger the fetching when called.

bafonins, 2020-01-28:
Please take into account that sorting table entries also triggers the entity title section to reload. I think this shouldnt be the case

kschiffer, 2020-02-05:
I'd suggest the following:
- Introduce new actions to retrieve the total number of entities per entity type (e.g. `GET_TOTAL_APPLICATION_COUNT`)
  - This will basically be a simple `List` call without filter parameters, from which we just extract the `X-Total-Count`-header
  - Store the result in a dedicated place in the store (I'd suggest e.g. `applications/totalCount`)
- Use these actions to fetch the total number of entities for the entity title section
- A distinction between `LIST_*` and `SEARCH_*` actions are then not needed (yet?) and we can avoid a larger store refactor

pgalic96, 2020-03-25:
Can we close this issue until (if) the distinction between list and search action becomes necessary again?

johanstokking, 2020-03-25:
Let's move to Backlog then

ryaplots, 2023-03-08:
Has something changed here? Do we now need the distinction between `LIST_*` and `SEARCH_* `. I just finished implementing the new total counts actions, so I could already also include that. @kschiffer

kschiffer, 2023-03-08:
I'll need to revisit the exact root cause and issues this produces. Let's keep this in the backlog until then.
What exactly are you working on with the total counts?

ryaplots, 2023-03-08:
I've introduced these this morning [https://github.com/TheThingsNetwork/lorawan-stack/pull/6096](https://github.com/TheThingsNetwork/lorawan-stack/pull/6096):
 
> * Introduce new actions to retrieve the total number of entities per entity type (e.g. `GET_TOTAL_APPLICATION_COUNT`)
>
>       * This will basically be a simple `List` call without filter [...]