TheThingsNetwork/lorawan-stack #1911: open cert.pem: no such file or directory (self-signed certificate)

zamashal, 2020-01-24:
#### Summary

After running docker-compose up, I am seeing the following error:
```
stack_1      | error:cmd/internal/shared:initialize_gateway_server (could not initialize Gateway Server)
stack_1      | --- error:pkg/gatewayserver:listen_frontend (failed to start frontend listener `MQTT/tls` on address `:8882`)
stack_1      |     address=:8882
stack_1      |     protocol=MQTT/tls
stack_1      | --- open cert.pem: no such file or directory
ttn_stack_1 exited with code 255
```
I made sure the ``cert.pem`` file exists in the same directory where I have ``docker-compose.yml`` and run the ``docker-compose up`` command from (**I am using a self-signed certificate**). I also made sure I have run ``chown 866:866 ./cert.pem ./key.pem``. Even when I changed it to an absolute file path in the ``docker-compose.yml``, I still got the same error.

#### Steps to Reproduce

1. Use the attached [docker-compose.yml.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/4109386/docker-compose.yml.zip) file
2. Follow the instructions on [Running The Things Stack
](https://thethingsstack.io/v3.4.2/guides/getting-started/running-the-stack/)
3. Check the log results of ``docker-compose up``

#### What do you see now?

```
stack_1      | error:cmd/internal/shared:initialize_gateway_server (could not initialize Gateway Server)
stack_1      | --- error:pkg/gatewayserver:listen_frontend (failed to start frontend listener `MQTT/tls` on address `:8882`)
stack_1      |     address=:8882
stack_1      |     protocol=MQTT/tls
stack_1      | --- open cert.pem: no such file or directory
ttn_stack_1 exited with code 255
```

#### What do you want to see instead?

It should run without errors or at least output a more relative error than "no such file" because the file definitely exists and has the suggested owner user and group (866).

#### Environment

VirtualBox VM (Ubuntu 18.04).

#### How do you propose to implement this?

It should be able to recognize the cert.pem file even if it's a self-signed certificate. The error should be more clear than just saying "no such file".

#### Can you do this yourself and submit a Pull Request?

Sorry I wasn't able to figure out the root cause of this error.

zamashal, 2020-01-24:
When I run with ``ttn-lw-stack start`` instead of ``docker-compose up``, I see: ``open cert.pem: permission denied``

johanstokking, 2020-01-24:
Hi @zamashal, great to see that you're setting The Things Stack up yourself.

Please see https://thethingsstack.io/v3.5.0/guides/getting-started/certificates/ for permissions for the certificates.

Also, see https://thethingsstack.io/v3.5.0/guides/getting-started/configuration/ for configuration of the Docker Compose file. Note that you need to have a volume mount for your certificates, so that they are available in the container. The mounts in the documentation are for ACME, but if you're using self-signed certs, you need to replace that mount by a local folder mount where your certs reside. Please see https://docs.docker.com/compose/compose-file/#volume-configuration-reference for more information.

johanstokking, 2020-01-28:
Seeing the warm emoji, I think this resolved your issue, hence I'm closing this. Please reopen if this didn't resolve your issue.

closed 2020-01-28 by johanstokking