TheThingsNetwork/lorawan-stack #1920: Resent email address verification mails?

bc547, 2020-01-27:
#### Summary
The lorawan-stack implements an email verification feature to make sure users have a valid email address. However, if this verification mail gets lost somehow, there is no possibility to send another one and manual intervention of an admin is required. 

#### Why do we need this?
In a setup with a lot of users, it would be handy if the user itself could request a new verification mail to avoid manual intervention of an admin via CLI.

#### What is already there? What do you see now?
I think most of the requested functionality is already in place since verification emails are already being sent.

#### What is missing? What do you want to see?
A way that no manual intervention whatsoever is needed for boarding new users.

#### Environment
lorawan-stack V3.5.0

#### How do you propose to implement this?
A simple button on the console (e.g. like forgot password) would help a lot?

#### Can you do this yourself and submit a Pull Request?
No

htdvisser, 2020-01-27:
Related: https://github.com/TheThingsNetwork/lorawan-stack/tree/feature/contact-info-cli

htdvisser, 2020-08-26:
CLI part was already merged in https://github.com/TheThingsNetwork/lorawan-stack/pull/2077, so what's left is the Console part.

NicolasMrad, 2022-03-29:
@kschiffer is this still relevant?