TheThingsNetwork/lorawan-stack #1924: Multiple scrollbars in desktop view

bafonins, 2020-01-27:
#### Summary

There are pages in the Console that have multiple scrollbars in the desktop view

#### Steps to Reproduce

1. Go to the pubsub create page
2. See 2 scrollbars that in certain cases make it problematic to scroll the page seamlessly

#### What do you see now?

2 scrollbars in the pubsub create page

#### What do you want to see instead?

just a single scrollbar

#### Environment

Google Chrome: Version 79.0.3945.117 (Official Build) (64-bit)
The Things Stack: v3.5.0

#### Can you do this yourself and submit a Pull Request?

yes, but i think it makes sense to add this fix into the responsiveness related prs

kschiffer, 2020-01-27:
This is related to this change:
https://github.com/TheThingsNetwork/lorawan-stack/commit/f5e5a3d4b62bda89a3f0ccef6bb6086200c8f2ac

Do you recall the context?

bafonins, 2020-09-21:
Looks like the issue is fixed

closed 2020-09-21 by bafonins